adafruit/Adafruit_NeoPixel #40: LED blinks and goes off immediately if setColor has two equal non-zero color components

peetonn, 2015-01-17:
__Description:__
In [strandtest.ino](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino#L26) with 1 pixel, if you try setting white color `colorWipe(strip.Color(127, 127, 127), 50);` pixel will never maintain solid color - it starts blinking.
__Expected:__
Pixel stays on.

jakobwenzel, 2015-02-02:
I can confirm this on a 16MHz Arduino with the 800KHz version.

tdicola, 2015-02-02:
Are you sure you have the ground from the LED strip connected to the Arduino ground?  A blinking light usually means there's a floating ground and the pixels are picking up noise as the signal.

jakobwenzel, 2015-02-02:
Other colors work fine.

If you only call .show() every second, you can see the LEDs flash very briefly. To reproduce this, take simple.ino, set color to 150,150,150 and delayval to something like 2000.
The data that is transmitted (as seen on my scope) seems to be correct. Only one packet is transmitted, so the LEDs do not seem to be turned off by the Arduino.

jakobwenzel, 2015-02-02:
I have tried with other LEDs. Some display the wrong behaviour for two equal values, some only behave incorrectly with all three values being the same, some behave correctly in all cases.

PaintYourDragon, 2015-02-02:
What are you using as pixels? Cut from a strip, sewable NeoPixels, bare WS2812, or other?

If bare WS2812: try adding 0.1 uF cap across + and -. You may be observing a PWM-induced brownout when the three elements switch on simultaneously.

jakobwenzel, 2015-02-02:
Correct guess! Adding a cap fixed my issue :)

closed 2015-02-02 by PaintYourDragon