adafruit/Adafruit_NeoPixel #48: Feature Request: updateLength()

davidhudman, 2015-06-15:
I'm trying (and failing) to make a function that works just like the updateLength() function in the Adafruit_WS2801 library (link below) that allows you to logically change the number of pixels in the strip. I think my attempt to add this failed because while even though all variables from the 2801 library are still present in the NeoPixel Library with the same names, some are now constant but weren't before. Please help :)

If you guys prefer for me to make this feature request in a different place, I can do that, too.

https://github.com/adafruit/Adafruit-WS2801-Library/blob/master/Adafruit_WS2801.cpp

davidhudman, 2015-06-15:
Feel free to make fun of me as this is my first time using Github. I made the change in my own repository and added that function, but I can't figure out how to let you guys know that I've fixed the issue. I thought that I might need to create a pull request, but it doesn't give me an option to select my stuff.

PaintYourDragon, 2015-06-30:
Good point! Added in latest commit.

closed 2015-06-30 by PaintYourDragon

davidhudman, 2015-07-02:
I'm not sure what I did wrong, but my fix is messing up my timing on my strip after I change the number of lights. I'm still not sure what the problem is. It's probably best to roll back to your previous commit.

PaintYourDragon, 2015-07-02:
Which version is exhibiting the problem -- your revision, or the current Adafruit commit? If the former, try the most recent 'official' code. If the latter, can you provide a simple example that demonstrates the problem? I'm changing lengths here OK now.

davidhudman, 2015-07-05:
Looks great! Not sure what I did wrong in my revision. Thanks!

davidhudman, 2015-07-09:
I think I'm finding it again now. When I set the length of the strip to 100
lights in the strandtest example, I change it to where I am just running 5
colorWipe() functions, one after another. I also add a serial println()
function at the end of the colorWipe() function that prints millis(). After
480 seconds of real time, the serial monitor and the millis function reveal
that my Arduino micro seems to think that it has only been running for 472
seconds.

Not sure why this is happening. Thoughts?

davidhudman, 2015-07-09:
BTW, the problem persists in the latest Adafruit NeoPixel commit. What's
even more interesting is that Adafruit_ws2801 library does not have this
issue when I use it with 100 LEDs. It's the exact same code with just a
different strip and strip constructor in my code. I know the ws2801 has a
clock pin, but I wouldn't think that could affect the Arduino millis()
function.

davidhudman, 2015-07-10:
During my testing when I set the number of LEDs in the Adafruit_NeoPixel strip constructor to 100, I printed millis() to the serial monitor and discovered that it is almost exactly perceives 2 seconds in the real world to be 1 second in the Arduino world. After 39 minutes and 55 seconds (in the real world) of testing, the millis() function indicated that Arduino thought only 20 minutes and 0 seconds had passed.

It's perception of time also seems to change when the number of LEDs changes. When I set the number of LEDs to 10, millis() was accurate based on the few minutes of testing that I did however it's possible that it could lose a few seconds over an hour or so, but I didn't run tests for that long. Setting the number of LEDs to 30 seemed to keep pretty accurate time as well, but I will have to do more testing to verify that. When I set the number of LEDs to 1000, it kept time well, but it would not turn on any of the LEDs like it would when I set the strip length to 10, 30, and 100.

What's so strange is that this problem will not exist if I comment out the Adafruit_NeoPixel strip and use the Adafruit_WS2801 strip instead with 100 LEDs and the physical WS2801 strip.

It's certainly an interesting problem to have because explicitly setting the number of LEDs in the strip to different values appears to affect Arduino's internal clock.