adafruit/Adafruit_NeoPixel #49: Doesn't compile on ESP8266: 'GPIO_OUT_W1TS_ADDRESS' undeclared

TimeTravelingOwls, 2015-06-30:
Arduino 1.6.4 on Huzzah ESP8266, compile produces this error:
```
/Users/Family/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266.c: In function 'espShow':
/Users/Family/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266.c:53:20: error: 'GPIO_OUT_W1TS_ADDRESS' undeclared (first use in this function)
     GPIO_REG_WRITE(GPIO_OUT_W1TS_ADDRESS, pinMask);       // Set high
                    ^
/Users/Family/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266.c:53:20: note: each undeclared identifier is reported only once for each function it appears in
/Users/Family/Documents/Arduino/libraries/Adafruit_NeoPixel/esp8266.c:56:20: error: 'GPIO_OUT_W1TC_ADDRESS' undeclared (first use in this function)
     GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, pinMask);       // Set low
                    ^
Error compiling.```

NeoPixelBus compiles fine. Do I need an additional #include for ESP8266?

PaintYourDragon, 2015-06-30:
Use the installation sequence currently shown in the guide here:
https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide

If you have any other method of ESP8266 support installed, remove that first (including any prior Adafruit Huzzah ESP8266 package) -- we currently recommend using the "Generic ESP8266 Module" setting in the Board menu.

Compiles OK w/esp8266 1.6.4-673-g8cd3697 package in Arduino IDE 1.6.5. Let me know if you're still encountering problems...any additional details on your setup may be helpful.

TimeTravelingOwls, 2015-06-30:
> any additional details on your setup may be helpful.
I was running Arduino 1.6.4 and esp8266 1.6.5-804-g2d340c7 from the staging package - fails with both "Generic ESP8266 Module" and "Adafruit Huzzah ESP8266" selected. 

Compiles and runs beautifully with esp8266 1.6.4-673-g8cd3697. Probably an issue with the staged code - I'll close this issue and check it again on the next release. 

Thanks for showing the ESP8266 some library love - I really appreciate the effort.

closed 2015-06-30 by TimeTravelingOwls