adafruit/Adafruit_NeoPixel #54: numPixels = 100 causes timing issues

davidhudman, 2015-07-16:
During my testing when I set the number of LEDs in the Adafruit_NeoPixel strip constructor to 100, I printed millis() to the serial monitor and discovered that it almost exactly perceives 2 seconds in the real world to be 1 second in the Arduino world. After 39 minutes and 55 seconds (in the real world) of testing, the millis() function indicated that Arduino thought only 20 minutes and 0 seconds had passed.

It's perception of time also seems to change when the number of LEDs changes. When I set the number of LEDs to 10, millis() was accurate based on the few minutes of testing that I did however it's possible that it could lose a few seconds over an hour or so, but I didn't run tests for that long. Setting the number of LEDs to 30 seemed to keep pretty accurate time as well, but I will have to do more testing to verify that. When I set the number of LEDs to 1000, it kept time well, but it would not turn on any of the LEDs like it would when I set the strip length to 10, 30, and 100.

What's so strange is that this problem will not exist if I comment out the Adafruit_NeoPixel strip and use the Adafruit_WS2801 strip instead with 100 LEDs and the physical WS2801 strip. I am running tests on this library using my own programs, but if the WS2801 works, that would suggest that there is an issue with the NeoPixel library. When I used examples from this library like "strandtest" and added in something like "Serial.println(millis());" the timing issues were still there, but it may have been different from the 2:1 ratio that I discussed above.

It's certainly an interesting problem to have because explicitly setting the number of LEDs in the strip to different values appears to affect Arduino's internal clock.

PaintYourDragon, 2015-07-16:
This is a Known Thing™. The NeoPixel library has to disable all interrupts during a call to show() in order to maintain cycle-accurate timing for the LED data stream...ANY interruption will corrupt this. The millis() and micros() functions depend on a periodic timer interrupt behind the scenes. Since the interrupt is briefly disabled during each show(), the clock will lose time. The longer the strip (the longer the data stream), the more time is lost. About 30 microsecond per LED per show() call.

Option 1 is to factor the 30uS/LED/show into your timing calculations; e.g. with 100 LEDs, clock will slip by 3 milliseconds with each show().

Option 2 is more complex, to configure one of the timer peripherals as a stable time reference. Either issuing a periodic interrupt (e.g. 60/sec) which calls show(), or without an interrupt, just comparing the current counter value against a prior state to measure elapsed time.

closed 2015-07-16 by PaintYourDragon

davidhudman, 2015-07-16:
Would you happen to have any code samples that would demonstrate either of those two options?

For Option 1, I'm thinking something to the effect of:

```
strip.show();
microSecondAddon += numLEDs * 3;
while (microSecondAddon >= 1000)
{
     milliSecondAddon++;
     microSecondAddon -= 1000;
}
```

Then every call to millis() could be supplemented by using "millis() + milliSecondAddon".

I'm really not sure what you're talking about with option 2. Could you provide an example or further explanation?

PaintYourDragon, 2015-07-16:
Option 1 looks viable, yeah. And probably easier to implement.

Here's an example of Option 2:
https://learn.adafruit.com/3d-printed-daft-punk-helmet/code

davidhudman, 2015-07-16:
In my application, I can't really let timing be off more than a second every hour.

I read through that code from the Adafruit Learn link, but I'm still not sure where the relevant parts are. I saw something in the setup() function that looked like it might be what I'm looking for:

 // A ~100 Hz timer interrupt on Timer/Counter1 makes everything run
  // at regular intervals, regardless of current amount of motion.

I'm still not sure if the part below that is what I should be trying to incorporate or not.

rhettc, 2015-07-17:
Taking a swing at this. This is one part desire to help, and one part desire to learn. 

Yes that's what you should be trying to incorporate. 

```C++
void setup() {
 // STUFF REMOVED
 // A ~100 Hz timer interrupt on Timer/Counter1 makes everything run
  // at regular intervals, regardless of current amount of motion.

#if F_CPU == 16000000L
  clock_prescale_set(clock_div_1);
  TCCR1  = _BV(PWM1A) | _BV(CS13) | _BV(CS11) | _BV(CS10); // 1:1024 prescale
  OCR1C  = F_CPU / 1024 / 100 - 1;
#else
  TCCR1  = _BV(PWM1A) | _BV(CS13) | _BV(CS11); // 1:512 prescale
  OCR1C  = F_CPU / 512 / 100 - 1;
#endif
  GTCCR  = 0;          // No PWM out
  TIMSK |= _BV(TOIE1); // Enable overflow interrupt
}
 
ISR(TIMER1_OVF_vect) {
  }
```

I'm going to repeat the above code below with comments added. Please bear in mind I don't know what you are comfortable with and what you aren't.

```C++
void setup() {

 // A ~100 Hz timer interrupt on Timer/Counter1 makes everything run
  // at regular intervals, regardless of current amount of motion.

/* The "#" indicates a pre-processor directive, something that happens prior to
the "main compilation phase" It's a common way to manage feature support. 
#if is a way to conditionally change the code that gets compiled. The constant 
F_CPU is used to determine the clock frequency, we need to (pre)scale the 
timer differently if we have a 16Mhz frequency, the first branch of this #if 
statement deals with that. */
#if F_CPU == 16000000L

/* If we are on a 16MHZ chip we need to use this prescale macro. A faster clock 
requires different increments. Macros are also special tools to affect code prior 
to main compilaton. A macro is like a function to (for example) consistently use common 
programming idioms. They can take arguments just like functions: clock_div_1 
is the argument here. The macro will be evaluated and and add code to the 
program.*/
  clock_prescale_set(clock_div_1);
  // These two lines are explained in the else branch since they appear in both
  TCCR1  = _BV(PWM1A) | _BV(CS13) | _BV(CS11) | _BV(CS10); // 1:1024 prescale
  OCR1C  = F_CPU / 1024 / 100 - 1;ged
// If it's not a 16Mhz clock assume we have an 8Mhz clock
#else
/* TCCR1 is a constant that represents the "Timer/Counter1 Control Register".
 A special variable used to manage the behavior of the timer/counter. The _BV() 
macro, I guess, is a binary value macro, probably used to facilitate the using 
constants like PWWM1A with bitmask OR operator "|".  Since registers squeeze 
as much info as possible out memory, they are usually a combination of bits 
that collectively act as flags. Using the OR operator we can turn on all the bits 
common to PWM1A, CS13, CS11. I didn't look into what each of those 
constants mean because the comment tells us the intent. To liberally 
generalize, megahertz are way too fast of a frequency for our needs (and 
possibly result in too high of a value for our internal counters) so let's start by 
reducing the resolution by a ratio of 1:512. */
  TCCR1  = _BV(PWM1A) | _BV(CS13) | _BV(CS11); // 1:512 prescale
/* Now we setup the  Output Compare Register to define how often we notified, 
or how often or clock "ticks". I think we adjust that value down by one since we'll 
be notified right "after" our counter overflows */
  OCR1C  = F_CPU / 512 / 100 - 1;
#endif
  GTCCR  = 0;          // No PWM out
/* Now we want to setup our interrupt. We'll use the |= to turn on the flags we 
need while preserving any existing flags of value of the Timer/Counter1 
Interrupt Mask Register. As the comment implies, we want to be 
interrupted/notified when the timer overflows the value we designated with 
OCR1C */
  TIMSK |= _BV(TOIE1); // Enable overflow interrupt
}
 
/* Now that we have configured our interrupt's behavior, we need to define 
what should happen when the interrupt triggers. The ISR macro is used to 
so setup the declaration of that function. Just like the loop function, we put 
our code between the curly braces. */
ISR(TIMER1_OVF_vect) {
  }
```

davidhudman, 2015-07-17:
That's awesome! Thanks! Could it really just be as simple as adding that snippet to the setup() function and nothing else?

My solution may have required a lot more trial and error, but it seems to work. I would really like to see if yours works because it seems like it would be more reliable. Here is mine below:

```
Adafruit_NeoPixel strip = Adafruit_NeoPixel(numLEDS, dataPin, NEO_RGB + NEO_KHZ800); // Initialization of the Adafruit NeoPixel strip
long milliSecondAddon = 0;
double microSecondAddon = 0;

...
strip.show();              // refresh strip display
microMilliSecondAddon();   // always call this after strip.show()

....

// always called after strip.show()
void microMilliSecondAddon()
{
	microSecondAddon += strip.numPixels() * 20.12;		// 20 was too slow, 20.25 too fast; 20.12 might be a hair to fast (<1 sec per 20min) increasing the variable will make Arduino perceive time faster than it is actually happening (lights go faster)
	while (microSecondAddon >= 1000)
	{
		 milliSecondAddon++;
		 microSecondAddon -= 1000;
	}
}

// used now to replace a call to millis()
long myMillis()
{
	return millis() + milliSecondAddon;
}
```

rhettc, 2015-07-17:
I think so. You'd have to adjust it for the timing that you want, unless you also want a 100hz rate. I haven't used interrupts myself but there might be complications if your timing is too tight. In a situation where your interrupt fires again before your code in the interrupt handler completes, I think that would be troublesome.

What's the rate you want?

I agree, I think this is the more reliable way to go. Not only from what I've gleaned, but also that it's what Adafruit did in that example when they wanted reliable timing. It might take some tweaks but I think it's essentially what you need. Try it out let's see what happens.

rhettc, 2015-07-17:
Also these may be of interest:
This page in particular, but the whole guide is worth checking out
https://learn.adafruit.com/multi-tasking-the-arduino-part-2/timers

The guide above links to this guide. I've perused this one and it's my favorite. A good mix of fundamentals and practical stuff.
http://letsmakerobots.com/content/arduino-101-timers-and-interrupts

davidhudman, 2015-07-20:
I'm using the Arduino Micro, so it's a 16mhz chip.

I'm getting error messages like this
error: 'clock_div_1' was not declared in this scope
error: 'clock_prescale_set' was not declared in this scope
error: 'TCCR1' was not declared in this scope
error: 'PWM1A' was not declared in this scope
error: 'CS13' was not declared in this scope
error: 'TIMSK' was not declared in this scope

All I did was just paste that short snippet into my setup() function and then pasted the ISR function outside the setup() function down below, so it makes sense that some of the variables weren't declared and it doesn't know what the clock_prescale_set() function is. I just don't know how I'm supposed to declare them without knowing what their types are.

I haven't read the articles from your above post with the learn adfruit multitasking and letsmakerobots links. If I should read those first, let me know.

rhettc, 2015-07-21:
Might be a missing include file

davidhudman, 2015-07-21:
That's what I'm thinking but I'm not sure what to add. I've already got
Adafruit_Neopixel.

PaintYourDragon, 2015-07-21:
Since we're getting down to register-level access to configure the timer, the code becomes extremely dependent on the microcontroller used. The Daft Punk helmet project uses a Trinket, so that low-level code is tied to the ATtiny85 chip and won't work the same (if at all) on other chip variants.

If using anything with a 328P chip (Arduino Uno, Pro Trinket, etc.), search around for the Timer1 Arduino library. This helps in setting up periodic events without having to muck around with registers yourself, as it's a really hard thing to get right.

rhettc, 2015-07-21:
#include <avr/power.h>

rhettc, 2015-07-21:
https://github.com/PaulStoffregen/TimerOne

This library supports the ATmega32u4, the one that's in the Micro you mentioned. 

```
Timer1.initialize(microseconds);

Timer1.attachInterrupt(function);
// Run a function each time the timer period finishes. The function is run as an 
// interrupt, so special care is needed to share any variables beteen the 
// interrupt function and your main program.

rhettc, 2015-07-21:
Worth mentioning that the design of the dotStar strips seems to have been inspired in part by complexities like your issue. Those strips seem to be a lot more user friendly for your scenario. Not suggesting you run out and buy some, but something to keep in mind for the future.

davidhudman, 2015-07-22:
You nailed it! The Timer1 class worked like a charm! It's keeping time
perfectly for hours. Just downloaded the code from github and checked out
some of the code samples, which are pretty much the same on the Arduino
playground website as well.

rhettc, 2015-07-23:
:tada: :joy: Excellent! Really pleased to hear that!