adafruit/Adafruit_NeoPixel #59: setPixelColor add the ability to target LED index from the end using negatives.

t3chguy, 2015-08-08:
I feel as though using a negative n in setPixelColor should internally target that many LEDs from the end, it'd be very easy to implement and if anyone else feels this way then I'll go ahead and Pull Request this functionality.

Thanks

gknauf, 2015-12-07:
Is it not already easy enough to just do
```
strip.setPixelColor(strip.numPixels() - i, c)
```
I think the library should keep as small as possible, and with addding few bytes here and there at some point it will not fit any longer into smaller AVRs like ATtiny85 ...

PaintYourDragon, 2018-01-24:
This would break compatibility with existing sketches that rely on the clip-values-off-either-end-of-strip behavior (the Cylon glasses project comes to mind, but there's probably others).

If this behavior is needed in your application, I'd suggest implementing a wrapper function that looks at the pixel index and handles negative values similarly to gknauf's example above (except use '+' instead of '-', since the index is already negative).

closed 2018-01-24 by PaintYourDragon