adafruit/Adafruit_NeoPixel #79: flashing strip with esp-12

i-n-g-o, 2015-11-18:
Hi.

using current master i see flashes once and a while using a neopixel-strip with esp-12.
I use pin-2 to drive the strip, power it directly from a LiPo and use a quite big (620uF) capacitor to stabilize the voltage.

i-n-g-o, 2015-11-26:
ping

PaintYourDragon, 2015-11-26:
Can't reproduce issue. Tested strandtest on ESP-12 (Adafruit HUZZAH) pin 2, Arduino IDE 1.6.5 against both stable (1.6.5) and staging (2.0.0-rc1) versions of esp8266, 5V USB or 3.7V LiPoly.

xopr, 2015-12-22:
Are you using (hardware) Serial at the same time?
I've come across a similar issue when using NeoPixels, PN532 and Serial (for debugging).
It became stable after I added Serial.end() and Serial.begin( baudRate ) around the pixels.show() command.

i-n-g-o, 2016-01-15:
finally i came around to continue with this.

for testing i use the example `strandtest`, which does not use Serial.

but throwing in a `Serial.end();`does not improve the situation.
even turning off interrupts with `cli();` does not solve this issue.
still i get occasional flashes on the strip.

i tested with 3.7V directly from a LiPo and with wall-plugged 5V

i-n-g-o, 2016-01-15:
ah!
seems to have been a bad connection after all. ;)
testing for longer, but looks good.

thanks for the replies.

i-n-g-o, 2016-01-15:
runs perfectly :)

xopr, 2016-01-17:
Ah, nice!
The trouble I had was a weak power supply; actually, it ran from the FTDI adapter, so I'm glad it didn't break down on me. Disabling CPU load by disabling serial made te problem less apparent, so for other people running into these kind of problems: good and stable power supply is key (and good connection of course :) .

i-n-g-o, 2016-01-17:
right.
my bad connection was at the data-pin to the strip which i connected using a breadboard.
after soldering the connection it worked.

closed 2016-05-04 by ladyada