adafruit/Adafruit_NeoPixel #87: feature: add per pixel brightness capability

tablatronix, 2016-01-10:
It would be nice to set per pixel brightness via `setPixelBrightness` or `setPixelColor(...,brightness)`

damiangr, 2016-07-30:
I need this functionality too. In my project I have to store LED color and change only the state to on or off.

tablatronix, 2016-07-30:
as a workaround, You can set maxbrightness to 255 then control offsets using your own function

```C++
void setColorwBrightness(uint16_t n, uint8_t r, uint8_t g, uint8_t b, uint16_t brightness){
  strip.setPixelColor(n, (brightness*r/255) , (brightness*g/255), (brightness*b/255));
}
```

But you will lose your rgb values, so you must keep track of them yourself if you need them.

ladyada, 2016-07-30:
wont happen in this library as it would bloat RAM quite a bit! please feel free to fork :)

closed 2016-07-30 by ladyada