adafruit/Adafruit_NeoPixel #95: GPL v3 vs v2 license issue

bperrybap, 2016-03-22:
The files Adafruit_NeoPixel.cpp and Adafruit_NeoPixel.h were initially checked in Dec 12, 2012 as stating the license was GPL 3 or later.
The file COPYING file was initially checked in Dec 12, 2012 with the text from the GPL 3 license.
These files set the license of the library to GPL v3.

Those main library files still state that the license is GPL v3; however,
in Dec 2015, the COPYING file was modified to contain the GPL 2 license.

The license in COPYING (gpl v2) is now in conflict with the specified license in the main library files and the original license of the library (gpl v3).

I'm not sure you can change from v3 back to v2, but if it is possible, it would require that every single contributor agree to this re-licensing.
If the re-licensing from v3 back to v2 is not allowed, or if all parties have not agreed to this re-licensing, then the COPYING file should be restored back to GPL v3 to match the original licensing and the specified licensing in the main library files.

ladyada, 2016-03-22:
it should all be lgpl 3

closed 2016-03-22 by ladyada

SlySven, 2016-03-22:
Actually the original commits in 2012 you referred to were under _**L**GPL 3.0_ as ladyada states (not *GPL 3.0*) - unfortunately that commit in Dec 2015 https://github.com/adafruit/Adafruit_NeoPixel/commit/c52c1ae8b801f65967d9bd6a9da19c4447c150f4 did briefly set things to the *more* restrictive **GPL3.0** which has just been undone.  You are correct that anyone who submitted commits in the last couple of months may have thought they were contributing under those terms, but I am sure this can be resolved fairly straightforwardly.

*A passing innocent-bystander (but IANAL)*

ladyada, 2016-03-22:
the code cpp/h files themselves were always LGPL3

bperrybap, 2016-03-22:
I got it.
I don't see any issue either since it does look like the intent from the beginning was to be LGPL 3 (or newer)