adafruit/Adafruit_NeoPixel #96: Turn off the whole strand after each cycle

kpasick, 2016-03-26:
Is there any way we can make it so that each cycle turns off after it runs its course? Instead of just remaining lit as it waits for the next button push?

tinue, 2016-03-27:
What are you trying to achieve?

The "runColorCycle.py" is only an example to show how it works. If you you want something that runs one cycle and then waits for a button press, and if it should wait with the LEDs turned off, you would simply add a "strip.clearStrip()" after the line that triggers the one cycle.

If a light show always ends in black, I fear that would not look nice e.g. in the "rainbow" program, if you want to run the light show continuously. After every cycle there would be a brief black flicker.

Of course I could add a boolean "endInBlack" to the base class. Would this be what you are looking for?

kpasick, 2016-03-28:
Hi there! Thanks for your response. I'm trying to program my buttons to simply run a single cycle with each new click. I'm a masters student at the university of Waterloo, in English, and this is a project we are doing for class. Coding is not my strong suit, of course. 

I've thrown together a code that almost works, minus the part where they turn off after. 

I'm going to try adding that line you mentioned below, and see how that works. 

I really appreciate your response!

Kari

kpasick, 2016-03-28:
Hello again. My Flora is acting up and not receiving any of the code. I've tried to manually reset it and upload, but it doesn't seem to be working. Any tips?

Also I thought I'd attach a photo and description of our project just for clarification, if that helps you out. Your code has been so helpful to it!

The blue button will activate the rainbow sequence (which represents a soothing aesthetic). The red button will activate the theatre chase (or flashing) in red. After about 7-10 seconds, the lights turn off again. There are 60 LEDs on the neopixel strip and we've confirmed that each button works. The glass is frosted to hide the internal wires and the actual strip. 

It's representative of the relationship between art and technology, and points a finger at the trend of modern art being able to almost predict our desires and react accordingly. Goofy, I know. 

Thanks,
Kari

ladyada, 2016-03-28:
github issues are for bugfixes to the libraries. not tech support
tech support for adafruit products is only done in the forums
forums.adafruit.com

thx!

closed 2016-03-28 by ladyada