adafruit/Adafruit_NeoPixel #107: Library seems to be causing flashing colors in parts of the strip (video provided)

jadekler, 2016-07-11:
Video: https://www.youtube.com/watch?v=aQaF095z5wg

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  Adafruit Trinket Pro 5V 16MhZ

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.6.9

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Upload 'simple' example

ladyada, 2016-07-11:
this is not a code bug
this is a tech support question. contact the strip seller

closed 2016-07-11 by ladyada

jadekler, 2016-07-11:
@ladyada Actually it seems to have been that my ground connection was junk. Replacing some cables fixed the problem. Sorry for the bad issue was scratching my head for quite a while.