adafruit/Adafruit_NeoPixel #108: Using delay() is really bad practice...

T-vK, 2016-07-12:
I recently started to use this library and I'm really disappointed by the examples. They use the delay() function which is really bad practice imo. Because instead of loop+delay you can use static variables and repetitive function calling. That way you would be able to do useful stuff like listening to rotary switches or controlling other LED strips while also doing the same thing as before. 

What I'm talking about is that instead of code like this:
`uint32_t red = strip.Color(255, 0, 0);

void loop() {
  colorWipe_stupid(red);
}

void colorWipe_stupid(uint32_t color) {
  static uint8_t wait = 50;
  for(uint16_t currentLed=0; currentLed<strip.numPixels(); currentLed++) {
    strip.setPixelColor(currentLed, color);
    strip.show();
    delay(50);
  }
}`

you could use code like that:
`uint32_t red = strip.Color(255, 0, 0);

void loop() {
  colorWipe_smart(red);
}

void colorWipe_smart(uint32_t color) {
  static uint8_t wait = 50;
  static unsigned long lastUpdate = 0;
  static uint16_t currentLed = 0;
 
  unsigned long now = millis();
  if (now > lastUpdate+delay) {
    strip.setPixelColor(currentLed, color);
    strip.show();   
    currentLed = currentLed>strip.numPixels() ? 0 : currentLed+1;
    lastUpdate = now;
  }
}`

I'd really like to hear what the developers of this library think about that.

ladyada, 2016-07-12:
the examples are designed to be simple so people can follow them. 
you are free to submit advanced working examples if you like, the default examples wont be updated at this time. 
thx!

closed 2016-07-12 by ladyada

T-vK, 2016-07-13:
Okay I understand the examples are there so that people can understand how to use the library easily. But it would be incredibly helpful to have examples that you can simply add to existing projects.  
Because right now I have to write every single animation almost from scratch.  

Would you like me to make a PR adding some examples of the kind that I had in mind? I'm asking because I saw a lot of PRs being rejected here and it would be a shame to write the code for nothing.

bgat, 2016-07-13:
If you are talking about a reusable library of patterns that NeoPixel users
usually require, I think that should be in a standalone library that
NeoPixel cross-promotes. Otherwise, the library becomes a collection of
applications--- and that eventually leads to distracting clutter as people
argue over code formatting, build environments, language selection, RTOS
selection, architecture selection, packaging, etc. etc. etc.: Whatever
aspect you choose, someone out there has a religion based around it. :-)

By keeping your tutorials, modules, reusable patterns, etc. separated from
the NeoPixel library, you and your community get to call the shots on all
the above with much less interference. You are also strongly motivated to
stick with approaches that work beyond just one version of the NeoPixel
APIs, you end up thoroughly testing the utility of those APIs in a greater
variety of realistic use cases, and you reach conclusive findings that you
can feed back to the library developers. Everybody wins.

The examples that come with the library itself should focus on the library
APIs only, and really should be stripped of anything else so that they're
guaranteed to work with a minimum of maintenance. They aren't intended to
be programming examples, they're intended to expand on the API
documentation and also provide a couple of "does my hardware even work?"
programs. Less is more.

A PR to add a reference to your library in the NeoPixel docs is a much
easier sell to a much smaller audience.

Sorry for the lurk,

b.g.