adafruit/Adafruit_NeoPixel #113: Whole strip go OFF when same RGB values in last LED

Tmthetom, 2016-11-05:
Hello, i have problem with WS2812 strip.
When i set same RGB values (like 255,255,255, or 121,121,121), in last LED in strip, whole strip is OFF. I can change any color i want, but until i wont change last LED in strip into state, when RGB values are not same, problem remains. So whole strip can have random colors, everything is working, but if last LED is 141,141,141 whole strip is OFF, i change last color to 141,140,141, eveything is working. I was trying this even with more peaces and problem is still the same.

Simple example of code:

#include <Adafruit_NeoPixel.h>
#define PIN 6
Adafruit_NeoPixel strip = Adafruit_NeoPixel(2, PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
}

void loop(){

 ...  wating until data arrives from serial .....
  
  // This is OK
  changeColor(0, 12, 18, 15);
  changeColor(1, 25, 25, 24);

  // This is OK
  changeColor(0, 10, 10, 10);  // Here can be even same color
  changeColor(1, 25, 25, 24);

  // This is NOT OK
  changeColor(0, 12, 18, 15); 
  changeColor(1, 25, 25, 25);  // When same color here, not working.

  ... Other code
}

void changeColor(int n, int r, int g, int b){
  strip.setPixelColor(n, r, g, b);
  strip.show();
}

PaintYourDragon, 2016-11-05:
Can't reproduce. What hardware is this running on?

Tmthetom, 2016-11-05:
Arduino Uno... Also was trying FastLED library and the same problem, i think its maybe hardware problem, i have two led strips both have 2x LED. Wating for 1200x LED from China, so after arrive, i will let you know, if problem consist.

Also sending whole code iam using (sorry for different language comments, it was requirement from my friend). Code will allow you to control through serial. 
Change color command: xL,R,G,B
-> L = LED number
-> R = Red color value
-> G = Green color value
-> B = Blue color value
-> Example: x1,255,255,255
Show changed colors command: z
-> Will applicate changes

Code for Adafruit_NeoPixel.h library
[RGB_Reciever.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/573459/RGB_Reciever.zip)

Code for FastLED library
[RGB_Reciever_FastLED.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/573460/RGB_Reciever_FastLED.zip)

jbkuma, 2016-11-05:
Try using unsigned int instead.  I had this issue with a project I was
developing as well, and making the change solved my issues.

Tmthetom, 2016-11-05:
Hello, thank you, so i tried data types from Adafruit_NeoPixel.h
-> setPixelColor(uint16_t n, uint8_t r, uint8_t g, uint8_t b)
[RGB_Reciever_uint.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/573512/RGB_Reciever_uint.zip)

It did not helped, but thank you for advice ;)

Tmthetom, 2016-11-05:
Also, that two peaces iam using for testing are one of the first NeoPixels from Adafruit ever maded, maybe it should be issue that was in first samples. I will try some research with Logic analyzer.

PaintYourDragon, 2016-11-06:
By "first NeoPixel," do you mean the v1 Flora pixels, a small PCB with LED on one side and chip on the back? If so, use NEO_KHZ400 in the constructor.

Tmthetom, 2016-11-07:
I was told, that its one of the first products.
Here is one iam using: [https://www.tinkerforge.com/de/shop/media/catalog/product/cache/2/image/600x600/9df78eab33525d08d6e5fb8d27136e95/b/r/bricklet_led_strip_ws2812b_strip_element_600.jpg](url)

To be sure, i tried 400KHZ, but it did not helped.

pstray, 2018-07-06:
I get the same effekt here, when R=G, the ws2812b turns off.