adafruit/Adafruit_NeoPixel #115: Losing time in the show() updates

BsAtHome, 2016-11-13:
Hi,

The library's show() function runs with interrupts disabled. This is no problem for short strips, but as they get longer and longer, the internal timing will be off due to missed timer interrupts.

On a standard 16MHz Arduino (f.ex. uno or pro mini) you have timer0 to keep time running at a 1.024ms interval. When the strip contains 34 LEDs or more, then the interrupts are disabled for longer than 1.020 ms (8 * 3 * 1.25us * numLEDs, plus some overhead), and the millis() time will lag by a millisecond or more each time an update occurs. For most projects this would not be a problem, but when programming games, it becomes an issue and the interaction timing is thrown off (events come later than expected). Especially since I use strip-lengths of 60...250 LEDs with millisecond game timing and frequent updates.

An obvious solution would be to use multiple small-length strips, but that requires more pins and, more importantly, creates visual artefacts because the LEDs no longer update simultaneously.

A solution, which I have been using, is to compensate the internal timer counter with the lost time just before the interrupts are re-enabled, like in (bottom of show() function):
...
extern volatile unsigned long timer0_millis;
timer0_millis += numLEDs / 34;
interrupts();
...
This hard correction is not perfect, but keeps time much better and keeps the game better within real-time.

There are several problems here. The above correction is non-portable, does not take into account the fractional correction in the timer interrupt and additionally has an awkward division. It works satisfactory for the Arduinos I've been using, but a better thought is required to make it more general and portable. Any better thoughts and possible integration into the code-base would be welcomed.

--
Greetings Bertho

houston47d, 2017-02-02:
<p>That is an interesting problem. Not one I think the NeoPixel library could handle in a way that will work in general. Are you using Timer1 for anything? At 16-bits it would be able to keep time across the NeoPixel show and allow you to correctly compensate the milliseconds counter. Or, for that matter, replace the use of Timer0 with Timer1 for the keeping of time.</p>
<p>I have a sketch that was having trouble with the long interrupt disables, and I added a stub function with weak linkage that is called just after interrupts are disabled (with a true parameter) and just before interrupts are re-enabled (with a false parameter). The weak linkage allowed me to replace it in the particular sketch without affecting other users of the library. And the linker in-lines the function (whether using the library supplied or my sketch supplied version) so there is actually no overhead of a function call.</p>

BsAtHome, 2017-02-02:
The generic way is to compensate at least part of the lost time for those interrupts missed, and then cross fingers that you don't have any further trouble. The compensation can be improved if you look at timer0 before and after the pixel-update, together with the knowledge of the time an update takes, to get a better idea of the real interval and compensate accordingly. There will always be a small race here.

I use timer1 for hardware tone-generation because the long delays cause "clicks" when you use the Arduino tone() functionality.