adafruit/Adafruit_NeoPixel #118: Travis CI fails on master

BsAtHome, 2016-12-05:
The master branch fails the Travis CI build test. There are several issues regarding the RGBWstrandtest example. There are name-collisions that prevent the build to succeed.

PR #114 (HSV addition) now includes the fixes needed to make Travis CI tests to succeed again.

Even is the HSV addition is rejected, then the Travis fixes should be applied to make continuous integration tests useful again.

--
Greetings Bertho

BsAtHome, 2016-12-05:
BTW, this is also related to issue #100.

BsAtHome, 2017-03-18:
The latest merge from PR #119 fixed one part of the Travis CI problem. The one left is that the RGBWstrandtest example will not build on the trinket. Disabling that specific test, by creating the file "examples/RGBWstrandtest/.trinket.test.skip" will make Travis happy.

The problem with the trinket test is a #define of "BIN" name-collision between the avr library (avr/include/avr/iotnx5.h) and the arduino library (hardware/arduino/avr/cores/arduino/Print.h; cascaded from Stream.h). This one may be rather hard to fix in a good manner if the Stream code cannot be excluded from being required. Therefore, disabling the test may be the simplest option for now.

ladyada, 2017-03-21:
ok!

closed 2017-03-21 by ladyada

BsAtHome, 2017-03-22:
I don't quite understand the reply "ok!" and subsequently closing the bug while Travis still fails. Is the fix rejected or not acceptable? Or, the fix is acceptable, but was not yet committed (maybe by mistake)?

ladyada, 2017-03-22:
hiya we commited but didnt push til today - thankx!