adafruit/Adafruit_NeoPixel #122: member variable backlight is not initialized

glansberry, 2017-01-18:
When creating an instance of Adafruit_NeoPixel inside of another class, like this 

```
#ifndef _LED_PROCESS_H
#define _LED_PROCESS_H
#include <Adafruit_NeoPixel.h>

class LEDProcess {
public:
    LEDProcess(uint16_t num_leds, uint8_t led_pin):m_num_leds(num_leds) {
        m_ledctrl = new Adafruit_NeoPixel(num_leds, led_pin, NEO_GRB + NEO_KHZ800);
        m_ledctrl->begin();
    }

private:
    Adafruit_NeoPixel* m_ledctrl;
};
#endif
```

The brightness variable does not get initialized to zero, and can be something random.  The Adafruit_NeoPixel  constructor needs to explicitly initialize it to zero, like this:

```
Adafruit_NeoPixel::Adafruit_NeoPixel(uint16_t n, uint8_t p, uint8_t t) : numLEDs(n), numBytes(n * 3), pin(p), type(t), pixels(NULL)
#ifdef __AVR__
  ,port(portOutputRegister(digitalPinToPort(p))),
   pinMask(digitalPinToBitMask(p))
#endif
{
  if((pixels = (uint8_t *)malloc(numBytes))) {
    memset(pixels, 0, numBytes);
  }
  brightness = 0;
}

```

My workaround has been to add the line `m_ledctrl->setBrightness(255);` after the begin()

ladyada, 2017-01-19:
hiya, isnt brightness(0) set in both instantiators?

https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L54

glansberry, 2017-01-19:
> Adafruit_NeoPixel.cpp#L54

It sure looks like it is.   I though the version I was pulling from Arduino
was the most current, but it looks like now that I see the github repo,
these are newer.  I'll clone from here.   Thanks!

Geoff Lansberry

Engineering Guy
Kuvée, Inc
125 Kingston St., 3rd Floor
Boston, MA 02111
1-617-290-1118 (m)
geoff.lansberry (skype)
http://www.kuvee.com
<http://www.kuvee.com>

ladyada, 2017-10-22:
i think this got resolved? reopen if still an issue!

closed 2017-10-22 by ladyada