adafruit/Adafruit_NeoPixel #124: library does not support ESP32

marcmerlin, 2017-01-26:
It's a somewhat new architecture supported by the arduino IDE, but currently there is no code to support it.
I've also opened https://github.com/adafruit/Adafruit_NeoPixel/pull/123 to give a compile time error to the end user warning that things aren't going to work on processors that don't have custom code yet.

closed 2017-03-18 by ladyada

marcmerlin, 2017-03-19:
Thanks for the merge ladyada
For anyone finding this bug later: note that the code merged works, but if you call it many times, eventually the timing will be off once in a while. This seems to be because ESP32 isn't a realtime processor, so sometimes it does other things and delays the core running the arduino code.
An alternative to bit banging with timers that could be off, is this other driver: https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers 
it uses the RMT hardware component to do this in the background with perfect timing (but at the time I'm writing this, I'm having issues with it potentially interacting badly with I2C, read https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/issues/4 for status )

ladyada, 2017-03-19:
yah its not perfect but so far its been working ok for me! good enough for now :)