adafruit/Adafruit_NeoPixel #131: Remove use of delay()

sej7278, 2017-02-23:
I'm trying to make a project where I push a button and it changes the Neopixel ring mode.

So push button once and it runs colorWipe(), press it again and its runs theaterChase() and so on, all in a switch statement.

The problem is that the button presses aren't always getting picked up as the functions use delay().

Is there some way to replace delay() with millis() or somesuch, in a similar way we would with button debouncing?

All I came up with is recording millis() at the start of a function, then running a while loop until millis() was greater than the required interval+the start time, but that's essentially what delay() does I guess, as it has the same problems.

dniklaus, 2017-02-23:
Try to use this [Timer component](https://github.com/dniklaus/arduino-utils-timer).

pcarver, 2017-02-23:
What I did was just to implement a function (which I called idelay for interruptible delay) which records the time it was called at the beginning (using millis()) and then enters a while loop where it calls whatever polling functions I wrote until millis() > calledTime + desiredDelay. If one of the polling functions returned an event requiring action then my idelay function returns early.

Generally I use global variables to track state changes so the polling functions would just set a global variable with the desired state (e.g. mode=colorWipe or mode=theatreChase for your example) so that when the idelay function returns it doesn't pass any data back, but just gets you back to a top level loop that checks the global variable(s) to see what state it should be in.

sej7278, 2017-02-23:
i tried the colorWipe() example from #108 which works, but i couldn't apply the same logic to the more complicated functions like rainbow() due to the nested *for loops* - one to do the fading and another to apply it to each led, i ended up with either random coloured led's or epilepsy-enducing flashing!

echamberlain, 2017-03-15:
rainbow() without delay() would look like:

Before:
```
void rainbow(uint8_t wait) {
  uint16_t i, j;

  for(j=0; j<256; j++) {
    for(i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i+j) & 255));
    }
    strip.show();
    delay(wait);
  }
}
```

After:
```
void rainbow(uint8_t wait) {
  static unsigned long lastMillis = millis() - wait;  // Persists lastMillis across calls to rainbow() and causes code to execute the first time rainbow() is called.  0 could also be used as an initial value, if rainbow() doesn't need to execute the first time it is called.

  if (millis() - wait >= lastMillis) {
    static uint16_t j = 0;  // Persists j across calls to rainbow().
    uint16_t i;

    for(i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i+j) & 255));
    }
    strip.show();

    j = (j + 1) % 256; // Increments j to 255, then resets to 0.
  }
}
```

ladyada, 2017-03-15:
hiya, this isnt a library issue per se so we're going to close it

our guides on interrupts/timers may be helpful!
https://learn.adafruit.com/multi-tasking-the-arduino-part-1

thx :)

closed 2017-03-15 by ladyada