adafruit/Adafruit_NeoPixel #132: Crash on ESP8266 if number of pixels is too high

casparom, 2017-03-09:
Summary: I compiled and run the strandtest script on my ESP8266 board with the number of pixels changed from the default 60 to 300 and that caused the chip to do a soft reset every couple of seconds.

Steps to reproduce:
1. Open the strandtest script
2. Change the number of pixels from 60 to 300
3. Compile and upload the script to ESP8266
4. Start the ESP8266 (dont need to attach LEDs for reproducing)
RESULT: soft reset is performed every couple of seconds.

ladyada, 2017-03-09:
this is a side effect of the ESP8266 RTOS which is not open source. there is no fix :)

closed 2017-03-09 by ladyada

casparom, 2017-03-15:
Do you know what is causing this exactly or if there is a workaround?

ladyada, 2017-03-15:
there is no workaround, try dotstar LEDs :)

myst729, 2024-01-10:
> there is no workaround, try dotstar LEDs :)

So this lib can also drive [DotStar](https://github.com/adafruit/DotStar-2020-8x8-Matrix-PCB)? The board uses APA102 LEDs which were not mentioned.