adafruit/Adafruit_NeoPixel #136: Initiating dynamically creates a panic

WeeJeWel, 2017-05-08:
When I create an instance dynamically, e.g.:

```
Adafruit_NeoPixel strip;

setup() {
   int numPixels = EEPROM.read(0); // e.g. `90`
   strip = Adafruit_NeoPixel( numPixels, DATAPIN );
}

loop() {
   strip.setPixelColor( 50, color ); // Boom crash!
}
```

The stack trace I've got is:

```
>>>stack>>>
3ffffed0:  40225e01 00000000 00640100 00000002  
3ffffee0:  0000076d 00000001 3ffeb24c 3fff1f74  
3ffffef0:  000000f0 3ffebed0 3ffebedc 401073f8  
3fffff00:  00000000 3ffebf42 3ffebef2 3fff1f70  
3fffff10:  3ffeb24c 3fff1f74 3fff31ac 4023a848  
3fffff20:  00000001 3fff242c 3fff242c 3ffee870  
3fffff30:  00000000 3fff31ac 0000001c 3fff242c  
3fffff40:  3ffeb23e 00000000 3fff31ac 40239c51  
3fffff50:  0500a8c0 00000122 00000000 00000023  
3fffff60:  00000002 00000018 40211777 3ffed728  
3fffff70:  3ffeb218 3fffdcc0 3ffe9d48 3ffe9d48  
3fffff80:  402116ea 3ffed728 00000000 3fff2494  
3fffff90:  3fffdc80 00000000 3fff31ac 402257c7  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
```

The code works great when I use:

```
Adafruit_NeoPixel strip( numPixels, DATAPIN );
```

Before `setup()`

PaintYourDragon, 2017-05-08:
Can't reproduce; works OK here. More details about the board in use may be helpful.

WeeJeWel, 2017-05-08:
Hm, odd. I'm running on a Wemos D1 Mini (esp8266).

PaintYourDragon, 2017-05-08:
Ah! I had to think about this for a moment. I was testing on AVR, which doesn't much care if you've got a pointer going off into outer space.

C++ doesn't deal with memory management the way something like Python would...the Adafruit_NeoPixel declared in setup() is actually local, despite being assigned to the global 'strip'. The language just doesn't handle references 'intelligently' the same way.

Couple ways to deal with this. First, and probably easier, note that this:
`  Adafruit_NeoPixel strip;`
actually invokes the empty constructor; i.e. it's equivalent to:
`  Adafruit_NeoPixel strip = Adafruit_NeoPixel();`
which allocates an Adafruit_NeoPixel object and assigns most of its elements to 0/NULL/uninitialized. Okay then...
In setup(), other functions can be used to prime those elements:
`  strip.updateLength(90);`
`  strip.updateType(NEO_GBR);`
`  strip.setPin(6);`
Don't forget to:
`  strip.begin();`
The code in loop() should then run as expected.

The other route, requiring a possibly-annoying syntax, changes 'strip' to a pointer:
`Adafruit_NeoPixel *strip = NULL;`
In setup(), use the 'new' operator to allocate the object...you can pass in length and pin and such as you were doing before:
`  strip = new Adafruit_NeoPixel( numPixels, 6 );`
Being that it's allocated, not on the stack, this object is now persistent and global.
But, the syntax for working with it is a little nuisancey if bringing in existing functions, most of which use the '.' operator. It's now '->' like this:
`  strip->setPixelColor( 20, 0xFF0000 );`
`  strip->show();`

WeeJeWel, 2017-05-21:
Wow, thanks for the extensive explanation! I'm still learning C++, and this helps a load!

closed 2017-05-21 by WeeJeWel