adafruit/Adafruit_NeoPixel #137: Recent modification breaks my Arduino Micro

ynk, 2017-05-09:
- Arduino board: Arduino Micro

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.12

- https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L149

Back to 50L, problem fixed.

PaintYourDragon, 2017-05-11:
Tests fine here on Arduino Micro. Any other hardware details you can provide? Type/size/length of NeoPixels? Tested with strandtest or different code?

ynk, 2017-05-12:
Hardware is standard neopixel RGBW, WS2812 with 7 of them. I used carefully
crafted custom code, which is a bit heavy to share (i fill up 90% of the
arduino's memory once compiled)

I can try to isolate and reproduce during the weekend if needed.

J.

ladyada, 2017-10-22:
we couldn't replicate - maybe fixed? maybe something else? please reopen if you can post some code :)

closed 2017-10-22 by ladyada