adafruit/Adafruit_NeoPixel #142: RFC: Support for fixed-size `pixels` buffer to reduce code footprint

jynik, 2017-07-20:
I am opening this tracker item to inquire as to whether the change described below would be a welcome PR.

With a Digispark (attiny85) and avr-gcc/g++ 4.9.2, I've been running into code-size limitations when using both the NeoPixel library and the DigiCDC library. (Granted, the latter is the larger component.)

I found that by replacing the dynamic memory allocation of the `Adafruit_NeoPixel::pixels` buffer with a fixed-size buffer defined at compile-time, I was able to recover > 600 bytes, which helped me significantly.

Thus, I'm proposing the use of an opt-in macro to change the definition of the `pixels` buffer. A fixed-size buffer sufficed for my purposes, but perhaps this can be massaged to be a "max" size, with proper bounds checks.

A quick and dirty example patch is posed below. If this general idea is something other folks feel is useful, I would be willing to clean this up and submit a PR.

***Adafruit_NeoPixel.h***
~~~patch
--- Adafruit_NeoPixel.h      2017-07-19 21:46:24.229821992 -0500
+++ Adafruit_NeoPixel.h.orig 2017-07-19 23:47:45.131900840 -0500
@@ -52,7 +52,6 @@
   void
     begin(void),
     show(void),
     setPin(uint8_t p),
     setPixelColor(uint16_t n, uint8_t r, uint8_t g, uint8_t b),
     setPixelColor(uint16_t n, uint32_t c),
     setBrightness(uint8_t),
@@ -77,10 +76,16 @@
   uint8_t
     pin,           // Output pin number
     brightness,
-   *pixels,        // Holds LED color values (3 bytes each)
     rOffset,       // Index of red byte within each 3-byte pixel
     gOffset,       // Index of green byte
     bOffset;       // Index of blue byte
+
+#ifndef FIXED_PIXEL_COUNT
+  uint8_t *pixels; // Holds LED color values (3 bytes each)
+#else
+  uint8_t pixels[FIXED_PIXEL_COUNT * 3];
+#endif
+
   const uint8_t
     type;          // Pixel flags (400 vs 800 KHz, RGB vs GRB color)
   uint32_t

~~~

***Adafruit_NeoPixel.cpp***
~~~patch
--- Adafruit_NeoPixel.cpp.orig 2016-02-12 18:42:34.000000000 -0600
+++ Adafruit_NeoPixel.cpp      2017-07-19 23:45:11.667425290 -0500
@@ -33,16 +33,30 @@
 
 #include "Adafruit_NeoPixel.h"
 
-Adafruit_NeoPixel::Adafruit_NeoPixel(uint16_t n, uint8_t p, uint8_t t) : numLEDs(n), numBytes(n * 3), pin(p), pixels(NULL)
-  ,type(t), brightness(0), endTime(0)
+Adafruit_NeoPixel::Adafruit_NeoPixel(uint16_t n, uint8_t p, uint8_t t) :
+#ifndef FIXED_PIXEL_COUNT
+  numLEDs(n),
+#else
+  numLEDs(FIXED_PIXEL_COUNT),
+#endif
+  numBytes(n * 3), pin(p),
+#ifndef FIXED_PIXEL_COUNT
+  pixels(NULL),
+#endif
+  type(t), brightness(0), endTime(0)
 #ifdef __AVR__
   ,port(portOutputRegister(digitalPinToPort(p))),
    pinMask(digitalPinToBitMask(p))
 #endif
 {
+#ifndef FIXED_PIXEL_COUNT
   if((pixels = (uint8_t *)malloc(numBytes))) {
+#endif
     memset(pixels, 0, numBytes);
+#ifndef FIXED_PIXEL_COUNT
   }
+#endif
+
   if(t & NEO_GRB) { // GRB vs RGB; might add others if needed
     rOffset = 1;
     gOffset = 0;
@@ -56,11 +70,13 @@
     gOffset = 1;
     bOffset = 2;
   }
-  
+
 }
 
 Adafruit_NeoPixel::~Adafruit_NeoPixel() {
+#ifndef FIXED_PIXEL_COUNT
   if(pixels) free(pixels);
+#endif
   pinMode(pin, INPUT);
 }
 
@@ -804,18 +820,6 @@
   endTime = micros(); // Save EOD time for latch on next call
 }
 
 // Set pixel color from separate R,G,B components:
 void Adafruit_NeoPixel::setPixelColor(
  uint16_t n, uint8_t r, uint8_t g, uint8_t b) {
@@ -863,7 +867,7 @@
     // Out of bounds, return no color.
     return 0;
   }
-  uint8_t *p = &pixels[n * 3];
+  uint8_t *p = (uint8_t*) &pixels[n * 3];
   uint32_t c = ((uint32_t)p[rOffset] << 16) |
                ((uint32_t)p[gOffset] <<  8) |
                 (uint32_t)p[bOffset];
@@ -883,7 +887,7 @@
 // native format and is not translated here.  Application will need to be
 // aware whether pixels are RGB vs. GRB and handle colors appropriately.
 uint8_t *Adafruit_NeoPixel::getPixels(void) const {
-  return pixels;
+  return (uint8_t*) pixels;
 }
 
 uint16_t Adafruit_NeoPixel::numPixels(void) const {
~~~

ladyada, 2017-10-22:
honestly...we wouldn't pull such a modification. the attiny85 is a lovely chip but its so limited, our examples dont do much more than twidle some LEDs and read a button... with availablity of Trinket M0 & Gemma M0, it solves a lot of problems without creating new ones (which, such a PR would cause for sure)

closed 2017-10-22 by ladyada

jynik, 2017-10-22:
No worries. I suspected it might be more trouble than it's worth, but just figured I'd ask anyway. Cheers! :)

ladyada, 2017-10-22:
awsome, thank you! :)