adafruit/Adafruit_NeoPixel #146: Expose numLEDs property

mlewand, 2017-08-20:
It would be helpful to expose `numLEDs` property. Currently it's a protected, so it can not be later accessed by any other objects.

In my case I have created a LED controller class, which uses `Adafruit_NeoPixel` class for communication with the LED strip. Ideally I don't want to duplicate led counts all over the place, it should be kept in one place so that it is always in sync.

It would be nice to expose it so that other libs can check the number of controlled LEDs at runtime.

This feature can be implemented either by exposing the `numLEDs` property or adding a getter returning a const value, so that it can not be modified by other code.

If you'd like to see this enhancement, I'm happy create a pull request for that.

PaintYourDragon, 2017-08-20:
strip.numPixels() ?

mlewand, 2017-08-20:
Oh my, absolutely, sorry for missing that. Thanks!

closed 2017-08-20 by mlewand

PaintYourDragon, 2017-08-20:
No worries! 🐲