adafruit/Adafruit_NeoPixel #149: Library still not supporting new WS2811 chip?

petermichaux, 2017-09-04:
I have been working with WS2811 RGB pixels since last year. Recently I bought new pixels and they do not work with my Feather board running the Adafruit_NeoPixel library. In the Arduino editor, I updated the library from version 1.0.6 to 1.1.2 to capture this change from 50 microsecond to 300 microsecond latch.

https://github.com/adafruit/Adafruit_NeoPixel/commit/ec1614ad8f8ce54f5928cfa1e08e6e7b3b958faf

The new WS2811 RGB pixels still don't work. Do you think there are still something needing to change with the Adafruit_NeoPixel library or something else is wrong?

Thanks!

PaintYourDragon, 2017-09-04:
Might be logic-level related. Old pixels at 5V used to be compatible with 3.3V signal, newer pixels require either a level shifter or slightly lower voltage (e.g. 4.5V from three alkaline cells, or 3.7V straight off a LiPoly).

petermichaux, 2017-09-04:
Thank you for your response. I hope this will solve my problem.

So the logic signal coming out of pin 10 of the Feather board is 3.3V?

Do you have a link to a more detailed explanation of how to do the level shifter? I imagine I'll need to buy parts.

petermichaux, 2017-09-04:
@PaintYourDragon Thank you for your response. I hope this will solve my problem.

So the logic signal coming out of pin 10 of the Feather board is 3.3V?

Do you have a link to a more detailed explanation of how to do the level shifter? I imagine I'll need to buy parts.

PaintYourDragon, 2017-09-04:
Yes, all Feather boards have 3.3 Volt logic. Two common approaches to the 5V NeoPixel issue...
Use a logic level shifter as shown here: https://learn.adafruit.com/neopixel-levelshifter
Or the 'sacrificial pixel' technique: https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/

petermichaux, 2017-09-04:
Thank you for the suggestions and links.

I tried running my sketch with version 1.0.6 of the Adafruit_NeoPixel library on the Feather and pixels powered by three AAA batteries in series with a measured 4.15V. Everything magically started working!

It seems the new WS2811 is not compatible with the 3.3V logic the Feather emits.

I did not need to upgrade the library from 1.0.6 to 1.1.1 for things to start working. But I did test with version 1.0.6 and 1.1.2 and everything works with either version of the library as long as I use a lower voltage like you suggested.

Thanks again.

I'll purchase a logic level shifter and try the sacrificial pixel technique too.

closed 2017-09-04 by petermichaux