adafruit/Adafruit_NeoPixel #153: setPixelColor resets ESP8266

git-tiger, 2017-12-06:
I am using this library to control LED strip with WS2812B.
When I use setPixelColor, the MCU resets with exception 9 or 2.
Below is my code to change colors.
`
void process_leds(void)
{
	switch (rc_leds.type)
    {
		case LED_SOLID:
		{
			if ( strip_length > 0){
				for (uint16_t i=0;i<strip_leds.numPixels();i++)
					// strip_leds.setPixelColor(i, rc_leds.r, rc_leds.g, rc_leds.b);
				strip_leds.show();
				Serial.printf("LED: %d %d (%d, %d, %d)", rc_leds.type, strip_length, rc_leds.r, rc_leds.g, rc_leds.b);
				rc_leds.type = LED_COMPLETE;
			}
			break;
		}
	}
}
`

The error message is 
LED: 3 6 (0, 25, 0)
Exception (9):
epc1=0x40218ff4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x19000039 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

dhalbert, 2017-12-06:
The `setPixelColor()` line is commented out, so the `for` loop is really executing the `.show()` call multiple times. (It's always good to use curly braces in all cases to avoid this kind of problem.)

Could you repost in https://forums.adafruit.com and include your entire code? Since we can't see how `strip_leds` is set up, it's not clear what the problem is. Thanks.

git-tiger, 2017-12-06:
Yes, I took out the for loop.