adafruit/Adafruit_NeoPixel #155: Support for BBC micro:bit

a-t-1, 2018-01-22:
[Adafruit_NeoPixel.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/1650477/Adafruit_NeoPixel.zip)
- Board:  micro:bit
- Arduino IDE version:  1.8.3

Hi,

I couldn't find support in the Adafruit_NeoPixel library for the awesome BBC micro:bit. I did find the Microsoft PXT/neopixel (JavaScript?) sendBufferASM function that runs the neopixels on the make:code website. (https://github.com/Microsoft/pxt-neopixel/blob/master/sendbuffer.asm)

The attached .cpp has the sendBufferASM function ported to the arduino IDE asm syntax, inserted just before
#else // Other ARM architecture -- Presumed Arduino Due
starting with
#elif defined (_VARIANT_BBC_MICROBIT_)

The original sendBufferASM is commented out (for verification). Very ugly now. I'm happy to clean it up but I thought this might help someone with more advanced skills verify my tinkering.

Differences between the new code and sendBuffer
- the assignment of the bitmask, following
https://github.com/sandeepmistry/arduino-nRF5/blob/dc53980c8bac27898fca90d8ecb268e11111edc1/variants/BBCmicrobit/variant.cpp

- the assignment of the *reg (similar to the other ARM architecture), following
https://github.com/sandeepmistry/arduino-nRF5/blob/dc53980c8bac27898fca90d8ecb268e11111edc1/cores/nRF5/SDK/components/device/nrf51.h
and
http://www.iot-programmer.com/index.php/books/27-micro-bit-iot-in-c/chapters-micro-bit-iot-in-c/47-micro-bit-iot-in-c-fast-memory-mapped-gpio?showall=1

- lsrs, adds and subs in sendBufferASM were not supported; changed to lsr, add and sub (with no detrimental consequenses)

The ported code was tested with example "strandtest" and a 150 rgb led string on pins 1 and 2 of the micro:bit. 

a-t-1

ladyada, 2018-01-22:
hiya - this looks cool, can you send it up as a pull request? :)

a-t-1, 2018-01-22:
Thanks! Of course. (newbie at github)
I think it worked: https://github.com/adafruit/Adafruit_NeoPixel/pull/156 
Needless to say it relies on the adafruit micro:bit libraries:
https://learn.adafruit.com/use-micro-bit-with-arduino/adafruit-libraries
a-t-1

ladyada, 2018-01-22:
nice work! we'll check it out :)

closed 2018-01-22 by ladyada