adafruit/Adafruit_NeoPixel #158: Loss of Arduino/Genuino 101 Compability

nicoleise, 2018-01-26:
Hello, 

This is my first ever GitHub issue report, so if something is amiss, don't shoot me. :) 

Using the Arduino IDE 1.8.5 (Windows Store 1.8.10.0), the library Adafruit_NeoPixel.h does not compile in newer versions of the library, when target platform is Arduino/Genuino 101. Version 1.1.3 is the lastest version, that compiles. The error given is "Error compiling for board Arduino/Genuino 101" along with countless individual errors.

Steps to reproduce: 
 1) Open Arduino IDE 1.8.5 and set board type to Arduino / Genuino 101
 2) Sketch => Libraries => Manage Libraries
 3) Install latest version of Adafruit NeoPixel by Adafruit (version 1.1.5)
 4) File => Examples => AdaFruit NeoPixel => Strandtest
 5) Click Verify

Expected result: Code should compile

Observed result: Code does not compile, but instead gives countless lines of compiler errors relating to the included code in the library. 

Steps to troubleshoot: 
 6) Using the Library Manager, downgrade to 1.1.4. Restart IDE. Recompile. Notice problem persists. 
 7) Using the Library Manager, downgrade to 1.1.3. Restart IDE. Recompile. Notice problem disappeared.

The issue isn't limited to the example sketch. I am developing a sketch for a prototype of a product. This sketch produces the exact same errors in the compiler as the example. 

I chose the Arduino 101 due to the onboard BLE capability, not realising it's at end of life. After an initial Google search, I thought this was the reason for the library to not compile, however I then noticed on this GitHub page and in the readme.txt, that the Arduino 101 *is* a supported platform. 

My suggestion; either remove the Arduino 101 from list of supported platforms, or troubleshoot which changes occured in 1.1.3 => 1.1.4, that would have caused this. To me, this issue isn't a huge dealbreaker, since I'll be basing my product on different hardware (as this one is end of life), but would make you aware of the issue nonetheless. 

// Nicolai

AMoo-Miki, 2018-02-07:
1.1.4 and 1.1.5 releases are broken for ESP32 as well. [This commit](https://github.com/adafruit/Adafruit_NeoPixel/commit/c53d8aa910014457ae556a224e44348c8986df23) fixes that but is not a part of a release yet.

Manually changing the 2 lines on my end fixed the compile errors for me. See if that does the trick for you too.

PaintYourDragon, 2018-02-07:
Just made a new release (1.1.6) which should address this; both ESP32 and Arduino 101.

closed 2018-02-07 by PaintYourDragon