adafruit/Adafruit_NeoPixel #159: Wron color on first pixel with ESP32

AMoo-Miki, 2018-02-07:
- Arduino board:  **HUZZAH32 (ESP32)**

- IDE version:  **platformio 3.5.2a7**

I have 2 pixels _(Breadboard-friendly RGB Smart NeoPixel)_ connected in series to the ESP32. No matter what I do, the first led _(pixel 0)_ is green, and if I set a color to it, it shows up different.

I switched the LEDs physically and still _pixel 0_ is green. FastLED, however, doesn't have this problem and shows the colors correctly.

I have tried with the latest code from this library, as well as with 1.1.3 and 1.1.4 and have the same problem.

```
#include <Adafruit_NeoPixel.h>

Adafruit_NeoPixel pixels = Adafruit_NeoPixel(2, 21, NEO_GRB + NEO_KHZ800);

void setup() {
  pixels.begin();
  pixels.setPixelColor(0, pixels.Color(0, 0, 0));
  pixels.setPixelColor(1, pixels.Color(0, 0, 0));
  delay(10);
  pixels.show();
}

void loop() {}
```

sansillusion, 2018-02-10:
Did you get your pixels together ? Some pixels are RGB some are GRB.

AMoo-Miki, 2018-02-10:
Yeah. The 2 pixels are from the same "board". Also, whether I set the LEDs as A-B or B-A, which ever is 0, lights up green for (0, 0, 0).

lyusupov, 2018-02-10:
I can also confirm the issue for ESP32 - one green pixel.

AMoo-Miki, 2018-02-12:
@lyusupov, your change does make the first pixel go off for (0, 0, 0), but the first pixel still gets the wrong colors. For example (171, 0, 85), which is pink, shows up as white.

KilianB, 2018-03-01:
I encounter the same issue occasionally (rarely and not reproducible) using an esp8266 as well. Afterwards setting the color of individual pixel will overwrite the first green pixel just fine. 
I faced the issue a lot more severe when I tried to run the led strip without level shifting the data signal from the esp. Are you converting the output to the correct 5v voltage range?

tueddy, 2018-03-01:
I have replaced my library with https://github.com/Makuna/NeoPixelBus.git 
Without any hardware changes the neopixel works fine on my ESP-32 device

xvinny, 2019-01-31:
> I have replaced the library with https://github.com/Makuna/NeoPixelBus.git
> Without any hardware changes the neopixel works fine on my ESP-32 device

I liked the source code of this library. It is very robust and well done.
I'm gonna test it, because the SmartLeds library is causing me problems.

boonkey, 2019-07-22:
I also got this issue with NodeMCU ESP-WROOM-32 and Adafruit 8x8 RGB matrix. 
for me, every time I use the show() method, I get different corruptions. the first corruption is always green on led #0, but after that it's a random corruption. 

this issue is also the same for my NodeMCU ESP8266 V1.0 (ESP-12E)

@xvinny  @tueddy  your suggested library looks good, but I'm looking for one which can print ASCII strings, have you any recommendation for that? Thanks.

Example for code with corruption:
Setup:
`Adafruit_NeoMatrix matrix = Adafruit_NeoMatrix(8, 8, 1, 1, 33,
                                               NEO_MATRIX_TOP + NEO_MATRIX_RIGHT +
                                                   NEO_TILE_TOP + NEO_TILE_RIGHT +
                                                   NEO_MATRIX_COLUMNS + NEO_MATRIX_ZIGZAG +
                                                   NEO_KHZ800 + NEO_GRB);
 matrix.begin();
    matrix.setTextWrap(false);
    matrix.setBrightness(50);`

Loop:
`i = (i + 1) % 10;
    Serial.println(i);
    matrix.fillScreen(0);
    matrix.setCursor(2,0);
    matrix.print(i);
    matrix.show();
    delay(100);`

tttonyyy, 2019-09-25:
Might not be software/timing at all.  I have first pixel go green but it's from noise on the DIN signal -  I don't even need the ESP connected, can just touch the DIN with a finger and that's enough noise to make the first LED go green.  So you may find it was programmed correctly but noise flips it to green quite quickly.

I think this is why they recommend 470R series resistor on DIN (close to the neopixel end) and potentially 100k pulldown to GND (again at the neopixel end) to reduce signal noise.

AMoo-Miki, 2019-09-29:
@tttonyyy, that might not be the case with me though since switching to another library (without any physical change, not even movement) made it work fine.

nevada750club, 2020-01-25:
I have found a good solution: in my code I call "show()" twice and the problem is solved 99,9% for me (without this solution the first led wrong color always).
I hope this solution solve your problem also.
Let me know ;-)

example:

for (int i=0; i<NUMPIXELS; i++)
{
    leds.setPixelColor(i, iColor);
}
leds.show();
leds.show();

acturcato, 2020-01-28:
There is a "known timing problem" between Neopixel and ESP32.
See the comment of Mike (Adafruit Support) on:
https://forums.adafruit.com/viewtopic.php?f=24&t=137339

In my case, I have replaced the library with https://github.com/Makuna/NeoPixelBus.git
Without any hardware changes, my Neopixel Ring (24 leds) works fine on my ESP-32 device!!!

Giotoc, 2020-03-06:
I had the same problem with a NodeMCU-43S and a single pixel. I had the problem only when I tried to switch off the pixel with pixels.setPixelColor(0, pixels.Color(0,0,0)); all other values were working correctly. 
The "workaround" to duplicate pixels.show() works for me, I am not going to change the library for now

XFactHD, 2020-03-25:
I am having the same problem with the builtin pixel on the Adafruit Feather M4 Express.
The color is always wrong when calling setPixelColor() and show() or clear() and show() directly after calling begin().
The "fix" to duplicate pixel.show() works for me as well. Another solution I found is to add a delay of at least 40 microseconds betwenn begin() and show().

hznupeter, 2020-05-04:
The "fix" to duplicate pixel.show() works for me as well.

dr2okevin, 2020-06-13:
I Have the same Problem. duplicate pixel.show() works only half. The Pixel gets green and ~100ms later goes off. Using some animations that means a constant green flashing first pixel.

chinloon, 2020-07-14:
Mine seems to give problem when I use with pubsubclient.h.  Double show() gets the colour to be correct but get a flash of bright Green occasionally first.

lifankohome, 2020-09-15:
How to solve this problem? what's the reason of first light flashing green?

luiscamaral, 2020-09-22:
I am facing the same issue. Not sure how to solve it yet., and double call to pixel.show() during an animation didn't help much but to cause some flickering.
I notice that if I disable Wifi (WiFi.disconnect()) the issue is gone. So, I am suspecting it is a timing issue with the library.

lifankohome, 2020-09-22:
**I knew the reason: the first pulse is wrong.**
I try to control the led with a stm32f1xx using PWM with DMA, but same problem oocured, every led is good but the first blink green sometime. I check the time sequence found that first high pulse is too long and low pulse is too short. So  i deleted all PWM code and replaced with delay funciton, everything goes well. Here is my tme sequence with delay function:
[https://gitee.com/lifanko/Car/raw/delay/timeline.png](https://gitee.com/lifanko/Car/raw/delay/timeline.png)
And you guys could get all the code at [https://gitee.com/lifanko/Car/tree/delay/](https://gitee.com/lifanko/Car/tree/delay/)

luiscamaral, 2020-09-22:
I've confirmed on the oscilloscope today and you're right. When I turn-on the WiFi on the ESP-WROOM-32, for some reason the first pulse go wrong and kind of mess with everything after it. I am not sure if we can use timed interruption to modulate pulses in ESP32, but it might be a good implementation if it actually works. 

Looks like the current implementation for ESP32 is using [_CycleCounter register_](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/esp8266.c) like bellow:
`__asm__ __volatile__("rsr %0,ccount":"=a" (ccount));`

Any suggestions on how to better debug/fix this?

lifankohome, 2020-09-23:
Maybe we should focus on why **only the first pulse has problem**.
why disable WiFi could make first pulse be normal, first pulse is the most impossible to be influenced.

luiscamaral, 2020-09-24:
[image]

I agree. Probably they are different issues as I am still getting first LED with green color after a few tries. 
I can see the short pulse on the logic analyzer, but still it (the LA) can decode the LED correctly.

On the image we can see a sequence of 16 LEDs signals sent to the first LED (1) and what the second LED get from the first (2).

tablatronix, 2020-10-01:
Same problem, there is a timing issue with the first bit first pixel, everytime, this is not a random interrupt.
Seems to happen in certain instances, cannot narrow down what execution is causing it.

[image]

tablatronix, 2020-10-01:
Ok so I figured I can skew the start bit, so that whatever is wrong with getCycleCount is done, then the first bit might send correctly.

I delayed it half 1.5 periods
```C++
    while(((c = _getCycleCount()) - startTime) < period*1.5); // Wait for bit start (EXTRA WAIT)
```

Seems to be working for now, this might also explain why calling delay(1) before show(); fixes other interrupt issues

ladyada, 2020-10-30:
plesae try https://github.com/adafruit/Adafruit_NeoPixel/pull/253

PR #253 (2020-11-05): Add RMT support to ESP32
Replaces the ESP32's espShow function with one that uses the RMT (Remote Control) peripheral to send timed pulses on the Neopixel line. Using the RMT should result in less flaky pixel performance when used with long Neopixel strings or in applications with heavy use of interrupts. Tested on a HUZZAH32 Featherwing.

This PR *should* resolve #159 and #139. However, I don't have a long enough string of Neopixels to reliably reproduce those errors on the original code, so any extra testing would be greatly appreciated to confirm this fixes all instability. This code should [... cut]
closed 2020-11-05 by ladyada