adafruit/Adafruit_NeoPixel #161: ESP8266 F12 and WS2811 flicker issue

KochC, 2018-02-17:
Hi, I try to create a Wifi-Colorpicker for my ws2811 strip.

All in all, I use the follow code (just parts here), all works fine, but I get LED-flickering. I assume its an issue with the Wifi, but how do I solve this? I also tried the FastLED library, but I got more flickers with that one. So the Adafruit seems to be better but not perfect.
```
#ifdef __AVR__
  #include <avr/power.h>
#endif
#include <Adafruit_NeoPixel.h>

#define LED_PIN 13
#define LED_NUM 5

Adafruit_NeoPixel leds = Adafruit_NeoPixel(LED_NUM, LED_PIN, NEO_GRB);
int color[ ] = { 0, 0, 0 };

void setup(){
leds.begin( );
}

void loop(){
for( int i = 0; i < LED_NUM; i++ ){
    leds.setPixelColor( i, leds.Color( color[ 0 ], color[ 1 ], color[ 2 ] ) );
  }
  leds.show( );
  delay( 500 );
}
```

rivsc, 2018-12-21:
Hi, I have the same issue. How do you know this issue caused by wifi ?
I replaced ESP32 with an arduino, I still have this problem.

erdemarslan, 2021-01-18:
Hi, you have many reason with this problem. First problem is current problem. Recalculate your strips watt and check your power unit. Then if you use long strip (more then e few meters), give power to strip at begin and end. Always connect GND's between strip parts. Use true cable which can carry your current on it. And this is too important; WS2812B and W2811 data will trigger with 5V but your ESP board 3.3V tolerant. If you use a few neopixel leds with ESP8266, D1 Mini or ESP32, this is not problem but if you want to drive large amount of neopixel leds, ws2812B or Ws2811 not generate data signal correctly. Please add 74HCT125 or similar buffer/line driver ic (level shifter) to your circuit.

For more details please check https://learn.adafruit.com/neopixel-levelshifter & https://flashgamer.com/blog/comments/level-shifting-neopixels-for-particle-photon

Sorry my bad English :)