adafruit/Adafruit_NeoPixel #163: SoftwareSerial interferes with SK6821 LEDs

ednieuw, 2018-04-29:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

ednieuw, 2018-04-29:
When using SK6821 LEDs the  SoftwareSerial library stops the communication to the LEDs. 
When the SoftwareSerial library is removed from the program the LEDs works OK.
The old slibrary NewSoftwareSerial library also interferes with the communication. 
With the very old outdated library AFSoftSerial the LEDs  works OK,
For the second serial communication (to Bluetooth) I use pin BT_RX = 6,  BT_TX = 7. changing these  pin numbers do not restore communication to the SK6812 LEDs
The SK6812 LEDs are connected to pin 5.
I use a Arduino Nano 
When this line, SoftwareSerial Bluetooth(BT_RX, BT_TX);   , is removed the communication to the LEDs is working. 
I upgraded to the Arduino IDE 1.8.5

driverblock, 2018-05-02:
Please post a complete sketch which demonstrates the problem, so that we can try to reproduce the problem.

ednieuw, 2018-05-06:
[DriverBlock.zip](https://github.com/adafruit/Adafruit_NeoPixel/files/1977686/DriverBlock.zip)
**Problem**
After changing from the WS2812 library with WS2812 LEDs to AdaFruit Neopixels to use the SK6812 pixels two problems occurred.
1 The communication to the LEDs on pin 5 stopped when the Bluetooth connection to pin 6 and 7 was started with the SoftwareSerial library. 
The FAB_LED library gave the same problems but after changing the RX en TX to higher pin numbers 10 and 11 the communication to the LEDs was working again. But this library uses too much memory and also did not work on pins 6&7. To use other pin numbers did not solved the problem with the Neopixel library
2 When typing text in the serial monitor to let the program do some action the text was read properly in function void SerialCheck(void) but is not passed in a String to the processing function ReworkInputString(SerialString);
when at line 600 //  Serial.print("  Serialstring: "); Serial.println(SerialString);  was introduced the function works as expected.
After preparing a small extract program, for you to test, no problems were found. Everything works without problems.
I removed and added many functions but until now I cannot pinpoint the problem.  
An Arduino Nano and UNO behaves identically
I attached in the ZIP the small working script, the larger not working script and all the libraries from my Library folder

Thanks in advance
Ed

ednieuw, 2018-05-16:
After renaming the defines used in my program and names of function calls with  the program was rebuilt from the last working version until the communication error occurred.
I pin pointed the problem to the three dimensional bool array. 
When the array
bool digit[][3][5] = { 
                     { {1, 1, 1, 1, 1}, {1, 0, 0, 0, 1}, {1, 1, 1, 1, 1} },  //0
                     { {1, 0, 0, 0, 1}, {1, 1, 1, 1, 1}, {0, 0, 0, 0, 1} },  //1
                     { {1, 0, 1, 1, 1}, {1, 0, 1, 0, 1}, {1, 1, 1, 0, 1} },  //2
                     { {1, 0, 1, 0, 1}, {1, 0, 1, 0, 1}, {1, 1, 1, 1, 1} },  //3
                     { {1, 1, 1, 0, 0}, {0, 0, 1, 0, 0}, {1, 1, 1, 1, 1} },  //4
                     { {1, 1, 1, 0, 0}, {1, 0, 1, 0, 1}, {1, 0, 1, 1, 1} },  //5
                     { {1, 1, 1, 1, 1}, {0, 0, 1, 0, 1}, {0, 0, 1, 1, 1} },  //6
                     { {1, 1, 0, 0, 0}, {1, 0, 0, 0, 0}, {1, 1, 1, 1, 1} },  //7
                     { {1, 1, 1, 1, 1}, {1, 0, 1, 0, 1}, {1, 1, 1, 1, 1} },  //8
                     { {1, 1, 1, 0, 1}, {1, 0, 1, 0, 1}, {1, 1, 1, 1, 1} }   //9
                      };
Becomes larger than digit[4] no signals are send to the SK6812 LEDs. 
if( digit[0][3][5] ) { ....} was the statement that stopped the communication. when this line was removed signal were send to the LEDs. 
After moving this array into PROGMEM the problem is solved. 
Placing the array into PROGMEM is  a work around I can live with.
Hope you can find out what stopped communication.