adafruit/Adafruit_NeoPixel #172: It seems that the code in ``Adafruit_NeoPixel.cpp`` may be dependent on compiles options

bogado, 2018-08-11:
When compiling using platformio that allows you to tweak the compiler options it seems that the compilation breaks if the -Os is not used. It seems that the assembler code on lines (959)[https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L959] and (588)[https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L588] both end up in the linking stage. 

This happens when I remove the -Os and add -O2 to the compilation on avr-gcc tool chain that is distributed with platformio, at this moment this is ``avr-gcc (GCC) 4.9.2``.

```
$ platformio -f -c vim run --verbose | tail -n -1
/tmp/ccYgbgRp.s: Assembler messages:
/tmp/ccYgbgRp.s:1523: Error: symbol `head20' is already defined
/tmp/ccYgbgRp.s:1538: Error: symbol `nextbyte20' is already defined
/tmp/ccYgbgRp.s:1583: Error: symbol `head40' is already defined
/tmp/ccYgbgRp.s:1608: Error: symbol `nextbyte40' is already defined
lto-wrapper: avr-g++ returned 1 exit status
/home/bogado/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
*** [.pioenvs/protrinket5/firmware.elf] Error 1
============================================== [ERROR] Took 0.69 seconds ==============================================
Environment protrinket5	[ERROR]
============================================== [ERROR] Took 1.42 seconds ==============================================
Environment huzzah     	[SUCCESS]
```

The same code compiles fine with the huzzah esp8266 toolchain for the feather huzzah8266, even when using just the -O2 flag.

ladyada, 2019-01-26:
probably! we use arduino IDE - for other IDEs there may be differences. if you can submit a non-breaking PR for a platform.io fix, that would be awesome!

closed 2019-01-26 by ladyada

Laserdance100, 2025-01-14:
I just ran into this issue. This might be closed, but it is **NOT** solved.

This is fairely straight fwd. At high optimisation level, it copy part of the assambly functions used to make the impulsions for writing on the leds several times, and it copy the labels head20, nextbyte20, head40 and nextbyte40 several times, hence the error.

To avoid this, one has to use the notation %= of [GCC](https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html#AssemblerTemplate), which will render all copied labels unique.
I'll make the PR. 

I had a [peckled colors issue](https://learn.adafruit.com/adafruit-neopixel-uberguide/the-magic-of-neopixels), while doing a text scroll from right to left on a 5x8 RGBDuino shield, a step each 100ms. When the pixel were updated the "on pixels" did merge together for a while that was very perceptible. Adding a 10ms "all black" instant before each left shift didn't help. Using -O3 on the whole project **completely** solved the issue.

<details>
<summary>Here are some pictures, courtesy of my fairphone2 crap camera, rolling shutter effect included:</summary>
[image]
[image]
</details>