adafruit/Adafruit_NeoPixel #173: Namespacing.

bogado, 2018-08-11:
I was reading the issues and pull requests, I've seen that there are problems with name clashes. The library is in C++ and Arduino already supports C++11 that have support for namespaces. Is there any reason not to use namespaces?

I believe that many issues could be avoided by enclosing everything into a namespace like the following :

```c++
namespace Adafruit {
// This level is optional and debatable.
// It would help if all adafruit libs started to use the same namespace scheme.
namespace NeoPixel { 

… all the code is here 

}

}

// This brings only the name people have on their code to the root namespace.
using  Adafruit_NeoPixel = Adafrut::NeoPixel::NeoPixel;

```

Another thing that could help is to use less macros and more ``constexpr`` constants or functions. 

for instance :

```c++
static const constexpr uint8_t NEO_RGB = ((0 << 6) | (0 << 4) | (1 << 2) | (2));
```

instead of : 

```c++
#define NEO_RGB ((0 << 6) | (0 << 4) | (1 << 2) | (2))
```

has the advantage of isolating this definition in the context it's used (no need to undef later) and as a bonus it has a proper type attached to it and this makes it a little bit harder to be misused that the simple "text pasting" semantics of the pre-processor.

bogado, 2018-08-11:
Just a clarification that this is more like a question than a request. All those changes are simple to execute, but they do depend on programming style and even programming preferences.

There might be some technical difficulties that I'm not aware that could make such change prohibitive also.

ladyada, 2019-01-26:
if it ends up breaking 6 years worth of example code, then we won't do it without a really really good reason :)

closed 2019-01-26 by ladyada