adafruit/Adafruit_NeoPixel #183: Adafruit_NeoPixel library not working wiht ESP8266 when switched to FCPU 160MHz

ChrisMicro, 2018-11-06:
The Adafruit_NeoPixel library is working fine with the ESP8266 at normal speed of 80MHz.
When switched to the higher speed with

  WiFi.forceSleepBegin(); // turn of wifi to reduce power consumption
  delay(1);
  system_update_cpu_freq(160); // run MCU core with full speed

it is not working any more.
The NeoPixelBus library is working with the higher speed.

https://github.com/Makuna/NeoPixelBus

Would it be possible to add this feature to the Adafruit Lib?

ladyada, 2018-11-06:
if you submit a PR we'll take a look!

ChrisMicro, 2018-11-07:
Thank you for the response. 
I found the issue. The following setup is needed when changing the CPUFrequency in the program:

Arduino IDE => Tools => CPUFrequency=160MHz

The NeoPixelBus library seems to work without that when using half speed in the program.

closed 2018-11-07 by ChrisMicro