adafruit/Adafruit_NeoPixel #188: Feature Request: LPC1768 chipset

ziemowitp, 2019-01-17:
I'm attempting to compile Marlin Firmware on my MKS Sbase 1.3 board which is based on LPC1768.  I am unable to turn on Neopixel support because Adafruit_NeoPixel which doesn't have support built for LPC1768 platform.

andyman198, 2019-01-20:
+1, and I know there are others.

Additional Discussion on this from the Marlin issues page here: https://github.com/MarlinFirmware/Marlin/issues/12159 

This is one of the last things to finish the LPC1768 implementation AFAIK on Marlin. There are example projects that have made NeoPixels work that may give a head start:
https://github.com/adamgreen/NeoPixelTree
https://os.mbed.com/users/JacobBramley/code/PixelArray/

tharts, 2019-01-25:
The problem I see there is that these are all relying on importing mbed.h.
Bitbanging would be the better solution.
It should work somehow like this https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

ladyada, 2019-01-25:
PR's gladly accepted :) we don't own one of these chips

ziemowitp, 2019-02-06:
I would love to help but I don't know enough about hardware but it looks like it just needs a small section added:

Pio            *port;
^~~
.piolibdeps\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1736:19: error: 'port' was not declared in this scope
Pio            *port;
^~~~
Compiling .pioenvs\LPC1768\CMSIS\driver\lpc17xx_rit.c.o
.piolibdeps\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1736:19: note: suggested alternative: 'qsort'

Pio            *port;
^~~~
qsort
.piolibdeps\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1737:12: error: 'WoReg' does not name a type
volatile WoReg *portSet, *portClear, *timeValue, *timeReset;
^~~~~
.piolibdeps\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1740:3: error: 'pmc_set_writeprotect' was not declared in this scope
pmc_set_writeprotect(false);
^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pioenvs\LPC1768\CMSIS\driver\lpc17xx_rtc.c.o
*** [.pioenvs\LPC1768\libd8d\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp.o] Error 1
 [ERROR] Took 24.82 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 24.82 seconds

thinkyhead, 2019-02-25:
All of these things together should help me to add LPC support to the NeoPixel cpp file:

[ ] Having the LPC datasheet
[ ] Having the linked code and Adafruit NeoPixel library code as references.
[ ] Having some reference code that does similar bit-bashing
[ ] Knowing the proper pins to use for SBASE, ReARM, Smoothieboard, etc.

tharts, 2019-02-28:
Here's an example for bitbashing Neopixels with an Arduino. As far as I know LPC should have a delay_ns function.
https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

p3p, 2019-03-07:
I've added LPC176x NeoPixel support to Marlin with my [fork](https://github.com/p3p/Adafruit_NeoPixel) as the LPC176x framework Marlin uses is not mature enough to have this pushed upstream at the moment.

ladyada, 2019-03-09:
thanks - when you can submit a PR please do :)

ziemowitp, 2019-03-19:
I compiled Marlin with your library but I can't seem to get it working.  What pin did you use?

tharts, 2019-03-20:
I'm able to compile Marlin with Neopixels support working. Using the current 2.0.x branch.
I use pin P1_12 but any other free pin should do fine.
Thanks for the great work!!!

ziemowitp, 2019-03-21:
It works!  Thank you.  PlatformIO kept on replacing the library with one from the repo.