adafruit/Adafruit_NeoPixel #198: I'm attempting to compile Marlin Firmware on my Skr pro board which is based on STM32F4  I am unable to turn on Neopixel support because Adafruit_NeoPixel which doesn't have support built for STM32F4 platform.

ledfreak3d, 2019-07-30:
lin Firmware on my Skr pro board which is based on STM32F4  I am unable to turn on Neopixel support because Adafruit_NeoPixel which doesn't have support built for STM32F4 platform.

ladyada, 2019-07-30:
if you can find someone who is an STM32 developer we'd love to see support!

ledfreak3d, 2019-07-30:
Wow a reply from the Lady her self

PKav89, 2019-08-09:
Well, it does not sound easy. The WS2812 LEDs need a precise pulse length and can be controlled by STM32 (from best to worst):

- via SPI+DMA, on any SPI_MOSI pin

- via U(S)ART, on any U(S)ART_TX pin

- via any Timer interrupt (incl. Systick), on any pin

- (the hard&&nasty way) via disabling all interrupts and counting delays via NOPs, on any pin

And, in our case, there is no other option but the last one. :( So, someone must create a delay_ns function in time.h, just like for LPC1768, and then just copy/paste the LPC functions with almost no changes.

danym21, 2019-08-31:
can this be used also for the stm32F4?
https://github.com/bigtreetech/Adafruit_NeoPixel/tree/master
 	Adafruit_NeoPixel.cpp 	add stm32f1 	17 days ago

ledfreak3d, 2019-08-31:
Prety sure it can still need to test it

fpistm, 2019-09-05:
Hi,
If this can help, I've made a quick try using the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
As STM32F4 is a cortex-m4 it has the DWT cycle counter so I quickly add support and it works on a
Nucleo-F401RE:
https://github.com/fpistm/Adafruit_NeoPixel/tree/stm32_core

[image]

ledfreak3d, 2019-09-05:
Hi Frederic.

Ill test and give it a try thanks for the headsup

ledfreak3d, 2019-09-05:
Anything special needed to compile ?

Running the STM32F4 on a 3d printer board with marlin 2.0 on it

 

https://github.com/fpistm/Adafruit_NeoPixel/tree/stm32_core

 <https://user-images.githubusercontent.com/20641798/64353192-a09f9800-cffd-11e9-8ed2-39a723fb6f29.jpg>

fpistm, 2019-09-06:
> Anything special needed to compile ? Running the STM32F4 on a 3d printer board with marlin 2.0 on it

I've just used the Arduino_Core_STM32 and this library updated with my branch.
About Marlin, I never use it. I know several contributor has provide the support of this core in Marlin 2.0 but I do not follow this.

ledfreak3d, 2019-09-06:
Ill see if i can get it up and running ill report back if i do

Need to get me a stm32 dev board for testing

danym21, 2019-09-06:
I have tried to compile but I become a error 
`.pio\libdeps\BIGTREE_SKR_PRO\_Adafruit NeoPixel_\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\BIGTREE_SKR_PRO\_Adafruit NeoPixel_\Adafruit_NeoPixel.cpp:1206:11: error: 'dwt_getCycles' was not declared in this scope
     cyc = dwt_getCycles() + CYCLES_800;
           ^~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\_Adafruit NeoPixel_\Adafruit_NeoPixel.cpp:1224:11: error: 'dwt_getCycles' was not declared in this scope
     cyc = dwt_getCycles() + CYCLES_400;
           ^~~~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\lib7e6\_Adafruit NeoPixel_\Adafruit_NeoPixel.cpp.o] Error 1`

I have tried to look into and I think there is a the same function in the Marlin\src\HAL\shared\Delay.h
`FORCE_INLINE` volatile uint32_t getCycleCount() { return DWT->CYCCNT; }`

the function in Arduino_Core_STM32-master\cores\arduino\stm32\dwt.h:
`static inline uint32_t dwt_getCycles(void)
{
  return (DWT->CYCCNT);
};
`
but I´m not able to bring it to work :-(
changing the code from dwt_getCycles() to getCycleCount() is not working I think it must be included some how.

fpistm, 2019-09-09:
`dwt_getCycles()` in an API I've added to manage DWT across all STM32 series (when DWT is available):
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/dwt.h
So I guess you can replace by Marlin one or directly with registers access. Anyway ensure DWT has been enabled and init.

danym21, 2019-09-09:
many thx, I have seen this yesterday also but when I use  explicite the new STM32 package in platformio.ini 
"platform = ststm32@5.6.0" and 
"platform_packages = framework-arduinoststm32@1.6.0"
I become the "PortA not declared" errors like in this issue:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/17#issuecomment-513594153
or
https://community.platformio.org/t/stm32-platform-upgrade-to-v5-5-0-errors-porta-portb-undeclared/8797
I using the same board BIGTREETECH-SKR-PRO-V1.1 and would like to have Neopixel support working on it.
can you give me a hint why this happens or where I have to look to fix this issue?
the pinnames.h etc looks all good for me and the ports are declared.
can it be a problem in marlin self?
looks like the issue has as fix a workaround to use the old "platform = ststm32@5.4.3".
this fix/workaround will fix the compile error but give us not the new functions :-(

fpistm, 2019-09-09:
Unfortunately, I will not be able to help on this. I do not use PIO.

danym21, 2019-09-09:
which IDE/Compiler/OS do you use? maybe I can try the same way like you use ?

fpistm, 2019-09-09:
Arduino IDE with the Arduino_Core_STM32. :wink:
We provide a JSON file to install all packages resuired:
https://github.com/stm32duino/wiki/wiki/Getting-Started

ledfreak3d, 2019-09-09:
And for platformio its just a question of loading the updated library ?

danym21, 2019-09-09:
It looks like, regarding the function `dwt_getCycles()` does not exist in the old 1.5.0.
but now I´m in a circle I would test with the Arduino IDE, but marlin 2.0 does not support it anymore.
what I have seen now is that the board is missing in 1.6.0 under variants.
@fpistm can it be that this is the issue?

fpistm, 2019-09-09:
I proposed only my fork as an example.
The BIGTREETECH-SKR-PRO-V1.1 was never added in the official STM32 core. Maybe it was added in PIO thanks Marlin, I don't know.
As I said, simply replace the dwt_getCycles() by the register access: DWT->CYCCNT
That's all.

fpistm, 2019-09-12:
Hi,
I've updated my branch and submit a PR #206.
To be generic across all STM32 series, it now uses the SysTick.

danym21, 2019-09-12:
thats great, many thx. I will test when I´m back home.

ladyada, 2019-09-13:
merged!

closed 2019-09-13 by ladyada

danym21, 2019-09-13:
@fpistm many thx it is working great.