adafruit/Adafruit_NeoPixel #200: Neopixel in Marlin

h00etn, 2019-07-31:
I have Neopixel enabled in Marlin as follows:

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
#define NEOPIXEL_TYPE NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
#define NEOPIXEL_PIN 5 // LED driving pin on motherboard 4 => D4 (EXP2-5 on Printrboard) / 30 => PC7 (EXP3-13 on Rumba)
#define NEOPIXEL_PIXELS 3 // Number of LEDs in the strip
//#define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
#define NEOPIXEL_BRIGHTNESS 200 // Initial brightness (0-255)
#define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup
#endif

#if ENABLED(BLINKM) || ENABLED(RGB_LED) || ENABLED(RGBW_LED) || ENABLED(PCA9632) || ENABLED(NEOPIXEL_LED)
#define PRINTER_EVENT_LEDS
#endif

**When I use only one LED in a strip - everything works perfectly.** Color is changing when warmig up and can be controlled with Gcode M150.

**BUT, when I have 3 LED:s in a row only first LED is "controllable", the other two have other colors.** 

If I send M150 R200 the first goes RED, second goes blue and third goes OFF. 
If i send M150 B200, the first goes GREEN, second goes OFF and third goes GREEN. 
If I send M150 U200, the first goes GREEN, second goes RED and third goes BLUE

How can I have all three with the same color and change as the first one?

I have imported Adafruit Neopixel in the library and connected it to SERVO2 on my MKS Gen L board. It is connected to SERVO_PIN 5 (which corresponds to SERVO2 in pins.h.

ladyada, 2019-07-31:
you're using NEO_GRBW but you probably dont have RGBW LEDs

h00etn, 2019-07-31:
Well, the first LED goes WHITE when starting up and when not triggered manually by Gcode (M150). But perhaps it can do that anyway?

What should I change to go with RGB LED (if that is what I have) to try if it makes any difference?

ladyada, 2019-07-31:
you'd need to talk to whomever designed your hradware and firmware - we dont know anything about Marlin

ladyada, 2019-07-31:
@GMagician  maybe you have any idea?

GMagician, 2019-07-31:
Don't know how neo pixel works in Marlin, but better to post an issue in marlin repository

makermelissa, 2019-07-31:
It could be that only the first LED is being addressed. Maybe add a for loop to address them all?

GMagician, 2019-07-31:
in code there is a loop using:
`adaneo1.numPixels()`
but I don't see any code initializing it

GMagician, 2019-07-31:
but it's on AGCM4 library...need to check  MKS gen

GMagician, 2019-07-31:
found..it's done in adafruit library

GMagician, 2019-07-31:
What I can see in Marlin is that 
`Marlin_NeoPixel` class has a field `static` Adafruit_NeoPixel adaneo1. Such class can handle up to 2 Adafruit_NeoPixel instances.
Such Marlin class is initialized with `adaneo1(NEOPIXEL_PIXELS, NEOPIXEL_PIN, NEOPIXEL_TYPE + NEO_KHZ800)` (I think it call Adafruit method)

Marlin calls setcolor that is
```c++
void Marlin_NeoPixel::set_color(const uint32_t color) {
  for (uint16_t i = 0; i < pixels(); ++i) {
    #ifdef NEOPIXEL_BKGD_LED_INDEX
      if (i == NEOPIXEL_BKGD_LED_INDEX && color != 0x000000) {
        set_color_background();
        continue;
      }
    #endif
    set_pixel_color(i, color);
  }
  show();
}
```

`pixels()` is `adaneo1.numPixels()`

so it should loop all 3 leds...

pearlythepirate, 2019-10-23:
@h00etn Have you been able to resolve your issue? Currently troubleshooting and I seem to have the exact same problem you are describing.

pearlythepirate, 2019-10-24:
> Yes, my mistake - I had chosen the wrong kind in the row "#define NEOPIXEL_TYPE NEO_GRBW". With the LEDs I have it sould have been "#define NEOPIXEL_TYPE NEO_GRB"

Thanks I'll give that a try!

jechs83, 2020-07-05:
did anyone find the solution?????  i try everything and the leds not changing  color at the same time, it does ramdomly each led, im abut to put a resistor on the data line of the strip but my board is mks robin e3 and it has a special port neopixel,  it should work directly , any idea???

Arganaz, 2020-10-13:
try all 36 combinations in this:
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L79