adafruit/Adafruit_NeoPixel #204: Feature Request: Apollo3 support

csonsino, 2019-09-09:
Would be nice to have NeoPixel support for SparkFun Artemis based boards that use the Ambiq Apollo3.

csonsino, 2019-09-09:
I wrote a quick and dirty proof of concept on the `apollo3` branch at https://github.com/csonsino/Adafruit_NeoPixel, but it uses the nasty "noop" timings.  Will start looking at improving the timing mechanism with interrupts, DMA, etc, but if anyone out there wants to put something together then I'd be glad to test out code.

ladyada, 2019-09-09:
PR's welcome, even if they are inelegant to start :)

dhalbert, 2019-09-09:
There is an implementation in https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/neopixel_write/__init__.c that uses PWM peripherals to do the timing, at the expense of using quite a bit more RAM to generate the waveforms.

csonsino, 2019-09-09:
PR posted.

I have a project that I want to port to the Artemis Nano board, but I should be able to continue working on a better timing implementation in a few days.

ladyada, 2019-09-09:
nice!

afoxinsocks, 2020-01-04:
Hi,

Just wanted to check in on this, did you ever improve the timing implementation? I've been using your nop hack version for a few weeks and it does work, though I'd love to see this work completed and actually brought into the official NeoPixel Arduino Library.

csonsino, 2020-01-05:
I implemented a few different timing mechanisms (see my `apollo3_flash_delay` branch - https://github.com/csonsino/Adafruit_NeoPixel/tree/apollo3_flash_delay).  You can switch between the mechanisms by uncommenting the (one) desired `#define PIN_METHOD_XXX` towards the top of `apollo3.cpp`.  Something to mention is that for the `PIN_METHOD_CTIMER_PWM` mode, I didn't complete the mapping in `apollo3PadToTimer()` so only a couple of the pins will actually work.

Now for the bad news- all of my timing mechanisms suffer from the same flaw that the application basically cannot do anything other than service LEDs in the application loop.  If I try to do anything else (I really want BLE functionality as well as LED control), then the LED timing gets messed up and stops working.

There is a suggested implementation over in https://github.com/adafruit/Adafruit_NeoPixel/pull/205 to use SPI and MOSI that might resolve the CPU timing problem, but as far as I know nobody has had the time to work on it.