adafruit/Adafruit_NeoPixel #208: Cannot compile NEO_KHZ400 for Digispark

R3tr0BoiDX, 2019-10-22:
I cannot compile my code using the `NEO_KHZ400` argument for the Digispark. It works only with the `NEO_KHZ800` argument. It works perfect fine with the `NEO_KHZ400` argument, when I try to compile it for my Arduino Uno. I'm running the latest Arduino IDE version (1.8.10) on Windows 10 and have the latest version of the NeoPixel.

I also wrote a [forum post](https://forum.arduino.cc/index.php?topic=642175.msg4345023#msg4345023) earlier on the Arduino forum, since I thought it would be my failure, but it seems, something is wrong.

ladyada, 2019-10-22:
hiya we dont really support 400khz anymore, fi you can fix the bug and submit a PR we'll review but we have neither 400khz pixels nor a digispark to test with :)

R3tr0BoiDX, 2019-10-29:
okay, I will give it a try, thank you!

pinpong, 2019-11-12:
It works with `NEO_KHZ400`!

ladyada, 2019-11-12:
great, then this can be closed?

R3tr0BoiDX, 2019-11-13:
> It works with `NEO_KHZ400`!

Well, yes. On the Arduino. Not on the Digispark. Anyway, I use the `NEO_KHZ` now anyway and it works as well.

closed 2020-10-30 by ladyada