adafruit/Adafruit_NeoPixel #217: Arduino Nano 33 BLE - LED Strip doesn't light up

misosec, 2019-11-22:
- Arduino Nano 33 BLE
- Arduino IDE version:  1.8.10
- NeoPixel library version: 1.3.2
- Sketch used: RGBWstrandtest.ino

 I have connected Nano 33 BLE with 74AHCT125 level shifter, but LED strip do not light up... but with regular Nano, strip works as expected without any changes to my connection scheme. 
No compilation errors.

Arduino forum post about this issue: [https://forum.arduino.cc/index.php?topic=649037.0](url)

ladyada, 2019-11-22:
hiya tech support is handled in the forums - please open an issue if you find a bug in the core library code.

closed 2019-11-22 by ladyada

misosec, 2019-11-28:
Hello,
but I did find a bug, didn't i? The biggest bug of them all...  _It-does-not-work_ bug.  It's pretty hard to catch this one... :-)