adafruit/Adafruit_NeoPixel #219: ColorHSV() function causes compiler error

systembolaget, 2019-12-26:
Arduino: 1.8.10 (Mac OS X)
Board: Adafruit Metro Mini

With this line "uint32_t alertColour = strip.ColorHSV(32768);" in my code, as per your instructions https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library-use even with the Adafruit NeoPixel library included, I get the compiler error:

'class Adafruit_NeoPixel' has no member named 'ColorHSV'; did you mean 'Color'?

PaintYourDragon, 2019-12-26:
Function’s definitely in there as of April 4, version 1.2.0.

Update Adafruit_NeoPixel to the latest via the Arduino Library Manager. If you sometimes use Teensy and installed that library via the Teensyduino installer instead…remove that version first, it was forked from an earlier NeoPixel release that didn’t yet include the ColorHSV() function.

systembolaget, 2019-12-27:
Thanks, that helped.

closed 2019-12-27 by systembolaget