adafruit/Adafruit_NeoPixel #221: GPIO16 cannot be used

sascha432, 2020-01-29:
I designed some PCBs with ESP8266/GPIO16 connected to a couple WS2812 and was kinda surprised that this port isn't supported and this isn't mentioned anywhere.
The FastLED ESP8266 page actually writes that this port is supported but it does not work there either.

I modified esp8266.c and made a version that is compatible with the Adafruit NeoPixel library.

https://github.com/sascha432/esp8266-kfc-fw/blob/master/lib/KFCBaseLibrary/src/NeoPixel_esp.c
https://github.com/sascha432/esp8266-kfc-fw/blob/master/lib/KFCBaseLibrary/include/NeoPixel_esp.h

Can be used standalone or replace esp8266.c

#include "NeoPixel_esp.h"

uint8_t pixels[numPixels * 8];
NeoPixel_fillColor(pixels, sizeof(pixels), 0xff0000);
NeoPixel_espShow(16, pixels, sizeof(pixels), true);

ladyada, 2020-01-29:
hiya if you can make it work please submit a PR :)

sascha432, 2020-02-25:
https://github.com/adafruit/Adafruit_NeoPixel/pull/222/files ;)

closed 2020-02-25 by sascha432