adafruit/Adafruit_NeoPixel #223: How to sync using custom for loop?

Berkays, 2020-02-10:
I want to use 1 byte per led in the strip because of the sram limit of CPU. Is it possible to allocated 1 byte per led at the expense of having only white color? How would i sync this byte array with the strip?
I think it should be possible to send same value 3 times for rgb values from the single byte per led array.

ladyada, 2020-02-10:
nope - not possible with the library, it would require a complete rewrite of the neopixel handling! :)

closed 2020-02-10 by ladyada

Berkays, 2020-02-10:
Why would it need a complete rewrite? I only have to send the array values 3 times doesnt it?

ladyada, 2020-02-10:
the code is completely unstructured for that kinda usage - you could fork and modify it for your own needs :)

Berkays, 2020-02-10:
I would try modifying it if assembly was not involved :)