adafruit/Adafruit_NeoPixel #224: Setting up brighness messes up color of the LED

xchg-dot-ca, 2020-02-20:
- Arduino board:  Adafruit CLUE
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.11
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
- Library version: 1.3.4

Setting brightness messes up with LED color, for example:

`#include <Adafruit_NeoPixel.h>
#define PIN        1 
#define NUMPIXELS 382
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);
#define DELAYVAL 500
void setup() {
  pixels.begin();
  pixels.setBrightness(10);
}

void loop() {
  pixels.clear();
  for(int i=0; i<NUMPIXELS; i=i+2) {
    pixels.setPixelColor(i, pixels.Color(150, 00, 0)); // ODD
    pixels.setPixelColor(i+1, pixels.Color(150, 22, 0)); // EVEN

    pixels.show();
    delay(DELAYVAL);
  }
}`

ODD and EVEN pixels will look the SAME despite different values of R and G are set.

afoxinsocks, 2020-02-20:
I'm pretty sure this is because setBrightness is forcing the maximum color value of each RGB LED within the NeoPixel, and not a "brightness" of a backlight or something like that. i.e., at low brightness (10 in your case), there are only 10 total levels of brightness for each color R, G, B. In your case, setBrightness(10) = just under 4% total brightness. And 22 * -.04 =  0.88, which in an integer is going to just get truncated to 0. You'd need to get to 25 to see the green LED come on when you have setBrightness(10).

Moral of the story: Make your brightness value higher or increase the amount of green you're requesting and then you'll see green come on. In other words, the lower your setBrightness value, the less fine resolution you have between RGB color blending.

xchg-dot-ca, 2020-02-20:
Probably this is documentation bug, documentation should state that what you just mentioned above.

caternuson, 2020-02-20:
There is a note about it being potentially lossy in the docs here:
http://adafruit.github.io/Adafruit_NeoPixel/html/class_adafruit___neo_pixel.html#aa05f7d37828c43fef00e6242dcf75959