adafruit/Adafruit_NeoPixel #227: Compiler warning ('pinMask' set but not used)

VanKurt, 2020-03-03:
Compiling with PlatformIO for ESP32 yields the following warning:

C:\Users\xxx\.platformio\lib\Adafruit NeoPixel_ID28\esp8266.c: In function 'espShow':
C:\Users\xxx\.platformio\lib\Adafruit NeoPixel_ID28\esp8266.c:35:51: warning: variable 'pinMask' set but not used [-Wunused-but-set-variable]
   uint32_t t, time0, time1, period, c, startTime, pinMask;
                                                   ^

ladyada, 2020-03-03:
yep there's warnings, ignore them :)

closed 2020-03-03 by ladyada

VanKurt, 2020-03-03:
But that's the only one coming up during the build process. Doesn't look too pretty ;-)

Also: it can easily be fixed. I've created a corresponding pull request:
https://github.com/adafruit/Adafruit_NeoPixel/pull/228