adafruit/Adafruit_NeoPixel #237: WS2815

Chipsian, 2020-07-17:
I have a probelm with the WS2815 and I don't know to solve this problem. I use the Arduino IDE and the Node MCU ESP8266. When I connect the WS2815 the problem is that every color is changing randomly only the first 3 to 5 LED and the rest is doing nothing but when I use the WS2812b everything works fine.  I have two WS2815 LED Stripes and both do the same weird thing.

ladyada, 2020-07-17:
if WS2812 works and WS2815 does not, sounds like that type of LED is not supported!

closed 2020-07-17 by ladyada

kefkevm, 2021-01-21:
Chipsian, i have the same problem, did you fixed it?

Reflective-Dimensions, 2021-01-21:
I didn't even start yet... keep getting sent wrong parts...then can't find anyway to get the wrong parts to work together. Won't even start to plug n play till I'm sure I won't damage the one part I have I do need that works.