adafruit/Adafruit_NeoPixel #244: Library doesn't compile on megaAVR-0 hardware when F_CPU != 16 MHz

MCUdude, 2020-08-12:
Originally reported in at https://github.com/MCUdude/MegaCoreX/issues/84.

For some reason, the library doesn't compile for any megaAVR-0 chip when compiling for a 16 MHz clock. Here's the error I'm getting when compiling for 8 MHz.

Any idea what may cause this?

```
Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:280:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTD |  pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:281:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTD & ~pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:387:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTB |  pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:388:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTB & ~pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:478:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTC |  pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:479:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTC & ~pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:568:18: error: no match for 'operator|' (operand types are 'PORT_t {aka PORT_struct}' and 'uint8_t {aka unsigned char}')
       hi = PORTF |  pinMask;
                  ^
/Users/hans/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp:569:18: error: no match for 'operator&' (operand types are 'PORT_t {aka PORT_struct}' and 'int')
       lo = PORTF & ~pinMask;
```

ladyada, 2020-08-13:
yah i guess your chip has a different way of managing PORT types, some work will be needed to add support for it!

MCUdude, 2020-08-13:
> yah i guess your chip has a different way of managing PORT types, some work will be needed to add support for it!

Is this library compatible with the Uno Wifi and Nano Every? The megaAVR-0's does have a different way of managing ports, yes.

ladyada, 2020-08-13:
somebody submitted a PR for the UNO Wifi i think, we didnt test it because we dont own one

MCUdude, 2020-08-13:
Ah, that explains it. The code in the PR was probably only changed for the 16 MHz part in the code, since that's what the official boards are running. However, the thing with MegaCoreX (and my other Arduino cores as well) is to leave the hardware settings up to the user, and not force a particular F_CPU, BOD level and all that.

petrkr, 2022-06-21:
Already 2 years and issue still persists. Do you have any workaround for this ?

ladyada, 2022-06-28:
we're waiting for someone to submit a patch - we dont own any megacorex boards :)

petrkr, 2022-06-28:
Bad is I do not have time to fiximg 3rd party libraries, if I will do, i will do only that :(

For me was solution use TinyNeoPixelx which is part of megacore Arduino sdk, only few fuctions are missing (like my loved rainbow), but port rainbow efect there is less work than refactor, understand and fix this huge library without breaking anything else.

ladyada, 2022-06-28:
sounds like you found a solution!

petrkr, 2022-06-28:
Yes, but I forgot it mention here as I am very busy and do lot of things 'at once'... so easy to forgot to mention everything...

If I will remember tomorrow, I will post link for alternative library and it could be solved as 'workaround' with some #pragma or something like 'use this instead' for mega core

petrkr, 2022-06-29:
Here is library, which works for megaAVR tiny cores.. Tested on attiny 1616

https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/libraries/tinyNeoPixel - since it is part of Arduino Framework for that board, so just simple replace
`#include <Adafruit_NeoPixel.h>`
for 
`#include <tinyNeoPixel.h>`

Most of API is compatible.