adafruit/Adafruit_NeoPixel #247: New Feature - Online Simulation for FastLED 😀 libraries

Puneethkumarsn, 2020-09-06:
Hi There

The libraries are very helpful for us. they are really good. We have been using the libraries from Adafruit for many of our simulations. 

We are working on an Arduino Simulator where we can see the output of the code visually. There are simulations already available for DHT Sensor, SSD1306 OLEDs, and more for Adafruit libraries.

What **we would like is to provide simulation pages for FastLED library** for the Adafruit page so that all the users of Adafruit products as well as new visitors can get a feel of how good the library is. **The user can verify the library online using the browser without having to install or download additional software**. This also enabled them to be 100% confident before they buy related project parts.

This will be a very good hint for the user - similar to "Try before you buy" 😀
Here is an example of what the user can see:
[image]

We hope it is interesting for you and helpful for the library users. We look forward to your reply,
We will share the link to the simulation pages of the FastLED library from Adfafruit.
Thank you

PS:
We have simulation pages already available for DHT and SSD1306 here:
https://wokwi.com/arduino/libraries/DHT-sensor-library
https://wokwi.com/arduino/libraries/Adafruit_SSD1306

ladyada, 2020-09-06:
this is the NeoPixel library not FastLED library?

Puneethkumarsn, 2020-09-07:
My bad. Definitely NeoPixel I meant 😅
Sorry.

Shall I share the simulation pages for NeoPiel Libraries?
We will be glad to do so.

ladyada, 2020-09-07:
sure!

Puneethkumarsn, 2020-09-07:
Thank you. Do you also have any feedback or suggestions to improve the simulator? Your feedback helps us to continuously improve the tool 🙏?

ladyada, 2020-09-07:
hi no feedback - please post url when ready to share!

closed 2020-09-07 by ladyada

Puneethkumarsn, 2020-09-10:
Hi there

Kindly find the link below: 
https://wokwi.com/arduino/libraries/Adafruit_NeoPixel

Kindly let us know once the link is put. We will be excited about that 😀

Thank you.

Best Regards
Puneeth

ladyada, 2020-09-10:
thanks added to https://learn.adafruit.com/adafruit-neopixel-uberguide/downloads

Puneethkumarsn, 2020-09-14:
Sorry for the late reply. Our Honour 😀