adafruit/Adafruit_NeoPixel #254: ESP32S2 divzero on v4.2 IDF

ladyada, 2020-11-12:
hey @hierophect can you try using Arduino IDE and the `v4.2 branch` of ESP32 BSP and see if you can figure out why running a neopixel demo on ESP32-S2 causes a fault? :)

note that you have to DISABLE PSRAM in the dropdown menu (you dont need it and it hardfaults on some other bug that isnt related)

```
.ESP-ROM:esp32s2-rc4-20191025
.Build:Oct 25 2019
.rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
.Saved PC:0x40025f15
.SPIWP:0xee
.mode:DIO, clock div:1
.load:0x3ffe6100,len:0x8
.load:0x3ffe6108,len:0x608
.load:0x4004c000,len:0xa38
.load:0x40050000,len:0x2848
.entry 0x4004c190
.Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
.
.Core  0 register dump:
.PC      : 0x4008df8e  PS      : 0x00060133  A0      : 0x800880bf  A1      : 0x3ffca9b0  
.A2      : 0x3ffbf6c8  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x0000001d  
.A6      : 0x00000001  A7      : 0x3f40908c  A8      : 0x3f413000  A9      : 0x04c4b400  
.A10     : 0x04c4b400  A11     : 0xfffffff0  A12     : 0x00000000  A13     : 0x00000000  
.A14     : 0x00000020  A15     : 0x0000001e  SAR     : 0x00000002  EXCCAUSE: 0x00000006  
.EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x40026ddd  
.
.Backtrace:0x4008df8b:0x3ffca9b0 0x400880bc:0x3ffca9d0 0x40081b1c:0x3ffcaa00 0x40081927:0x3ffcaa50 0x4008142f:0x3ffcaa70 0x4008275e:0x3ffcaa90
.
.
.ELF file SHA256: 0000000000000000
.
.Rebooting...
```

hierophect, 2020-11-13:
@ladyada sure, I'll look into it

me-no-dev, 2020-11-13:
You should try using [this](https://github.com/me-no-dev/EspExceptionDecoder) to see where the error happens. It might be in IDF itself, which will be fixed once I update the SDK.

hierophect, 2020-11-16:
@ladyada @me-no-dev Ok I got a little muddled up by the version numbers here so I want to make sure I understand this correctly. The current implementation of this library is written for ESP-IDF version 3.2 which is all that is supported in the latest BSP release that you get with the Arduino board manager. I had to make a number of changes to bring code over from Circuitpython, which is currently using a commit a little ways after 3.3, since a lot of RMT functions don't exist in 3.2.

If this update moves to IDF v4.2 I'm kind of surprised it even compiles - when I was moving between IDF versions in Circuitpython there were always quite a lot of breaking API changes. How would you want me to deal with versioning issues like this, since the 4.2 code isn't available yet from the Arduino IDE? Is there a good way of detecting what BSP version we're using inside the library itself so I can handle multiple IDF API versions at once?

ladyada, 2020-11-16:
@hierophect i think 4.2 is coming in the next release of ESP32 so we may need to have different files or ifdef'ing to determine which version
@me-no-dev whats the best way to determine the IDF version in Arduino?

me-no-dev, 2020-11-16:
Have you tried to use Arduino's own RMT API? It will not require you to check versions or worry if it's working 😄 

@ladyada [this header](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_idf_version.h) is included in Arduino starting with 1.0.5-rc1. You can use it to detect the version.

[ESP32 Arduino RMT API](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.h)

hierophect, 2020-11-16:
@me-no-dev I'll take a closer look at using it - I imagine I'll have to migrate from the ws2812_rmt_adapter that the IDF uses to a new approach.

me-no-dev, 2020-11-16:
This is an example with one LED. Maybe can help you :) 
```cpp
#define NR_OF_ALL_BITS 24
rmt_data_t led_data[NR_OF_ALL_BITS];
rmt_obj_t* rmt_send = NULL;

void setLedColor(uint8_t r, uint8_t g, uint8_t b){
    uint8_t col, bit;
    uint8_t i=0;
    uint8_t color[3];  // RGB value
    color[0] = r;
    color[1] = g;
    color[2] = b;
    for (col=0; col<3; col++ ) {
        for (bit=0; bit<8; bit++){
            if ( (color[col] & (1<<(7-bit)))) {
                led_data[i].level0 = 1;
                led_data[i].duration0 = 8;
                led_data[i].level1 = 0;
                led_data[i].duration1 = 4;
            } else {
                led_data[i].level0 = 1;
                led_data[i].duration0 = 4;
                led_data[i].level1 = 0;
                led_data[i].duration1 = 8;
            }
            i++;
        }
    }
    // Send the data
    rmtWrite(rmt_send, led_data, NR_OF_ALL_BITS);
}

void rmt_setup(){
    if ((rmt_send = rmtInit(PIXEL_PIN, true, RMT_MEM_64)) == NULL){
        Serial.printf("init sender failed\n");
    } else {
        rmtSetTick(rmt_send, 100);
        setLedColor(0,0,0);
    }
}
```

hierophect, 2020-11-18:
Apologies for the Arduino-noob question here, but what's the easiest way to get on the `v4.2` release of the ESP32 BSP? I assume it's swapping some Arduino system folder out for a clone of the git repo, but I can't seem to find the right folder in the package resources. I guess things have changed from when everything was in `hardware`.

If we swap to the RMT API, I have a couple options for how I could implement it:
1) I could stick with the current Circuitpython-based system, where every call to the ESP_Show function reserves and then un-reserves an RMT channel. This allows as many Neopixel objects as you want, since no channels are permanently reserved, but it's slower because it's running init and deinit every call. 
2) Instead, I could reserve an RMT object within the Neopixel object itself, and only release it in the destructor. This is fast, but limits the number of Neopixel objects you can have to 8 channels on the ESP32 and 4 on the ESP32-S2.
3) I could assume that Neopixel calls will always be made sequentially (or enforce it with interrupt protection), and reserve a single RMT object for all Neopixel objects in the first Init. We could make this object never be released, or alternatively track how many Neopixel objects we have in the program, and only release it when they're all deconstructed.

Any input on which of these might be the most appropriate?

ladyada, 2020-11-18:
heres' how to get the 4.2 branch
https://learn.adafruit.com/adafruit-metro-esp32-s2/using-with-arduino-ide

2. option is fine. folks do not need more than 4 or 8 and we'll document it

hierophect, 2020-11-23:
Unfortunately I can't say I'm having much success in integrating the RMT API into the library - it works ok as a sketch, but in the context of the library it no longer returns to a low idle level and has a bunch of unexpected pulses every 120ms which screw up the neopixels. I'm not sure why it breaks down like that exclusively in the library, I can spend more time tracking that down if desired.

@ladyada I am unable to replicate your crash on the Saola 1 Wrover or the Metro S2 (though I don't know how to get crash outputs on the latter), it seems to work ok for my basic testing sketch. If you pass me the one you've been using, I could try that instead.

ladyada, 2020-11-23:
@brentru are you having issues still? i cant quite replicate either