adafruit/Adafruit_NeoPixel #256: Adding / getting help with support for UCS8903 RGB driver IC

CreativeRobotics, 2020-11-27:
I was hoping to adapt this library for the UCS8903 IC - Its very similar to the WSC IC's IC used in neopixels, the code and timing for bits is nearly identical, but it uses 16 bits per pixel and each of the three channels can drive a string of 7 LED's from a 24V supply.

Applying this library directly to a string of them does appear to drive the IC correctly, but because of the pixel resolution difference the colors are garbled.

I'm planning to fork the library and try and edit it to work with this IC, it may end up as a fork that only works with this IC, but if possible, I would like to get it to work with neopixels as well and then issue a pull request ...

Do any of the people most familiar with this code know if altering the bit resolution for pixels is likely to be a quick and easy fix? - some pointers here might save me a lot of time digging though the code!

Thanks in advance.

ladyada, 2020-11-28:
you would likely not be able to change the timing to this library - it is tuned for NeoPixels and we have many thousands of people using it. you can fork if you need. check out fastLED as well

closed 2020-11-28 by ladyada

CreativeRobotics, 2020-12-05:
Ok. The timing is actually the same, but for the moment I just borrowed the SAMD21 section of code and got it working with 16 bit color values. Changing the entire library to allow for both 8 and 16 bit color values looks like a big job.