adafruit/Adafruit_NeoPixel #274: Add Support for porting the library to custom boards with generic GPIO Names (e.g. PJ3)

mnemocron, 2021-05-22:
The library only works with Arduino IDE "digital Pin Numbers".
I have a project with the `ATmega2560` where I use pins that are not exposed on the Arduino Mega board.
Therefore they are not defined as digital Pins in the IDE and not supported by the Adafruit NeoPixel Library.

### Example:

Only the last 3 Pin definitions work.

```cpp
#define LED_DATA_1 PJ6
#define LED_DATA_2 PJ5
#define LED_DATA_3 PJ4
#define LED_DATA_4 PJ3
#define LED_DATA_5 PJ2
#define LED_DATA_6 33 //PC4
#define LED_DATA_7 34 //PC3
#define LED_DATA_8 35 //PC2
```

I see that the FastLED Library supports adding new pin definitions.

https://github.com/FastLED/FastLED/blob/master/PORTING.md

I was able to get the LEDs working with their pin definition and just defining my own "digital Pins" starting at 201.

```cpp
//_FL_IO(J);
_FL_DEFPIN(201, 6, J);
_FL_DEFPIN(202, 5, J);
_FL_DEFPIN(203, 4, J);
_FL_DEFPIN(204, 3, J);
_FL_DEFPIN(205, 2, J);
#define HAS_HARDWARE_PIN_SUPPORT 1
```

However, the FastLED Library does not support RGBW and I was wondering if you could add this feature.

ladyada, 2021-05-22:
yeah we wouldnt do this, we found the fastLED method is annoying and requires changes/releases for each board. so we would recommend making a new board definition for your variant!

mnemocron, 2021-05-22:
🤔 I guess that could work. Thanks for the hint.

closed 2021-05-22 by mnemocron

ladyada, 2021-05-22:
good luck :)