adafruit/Adafruit_NeoPixel #276: Add support for ESP8266 Arduino Core 3.0.0

izbushka, 2021-06-11:
Espressif8266 3.0.0 release breaks the library.
It builds and even sometimes works (randomly setPixelColor sets a correct color), but in general leds are white.
My configuration is:
platform = espressif826@^3.0.0
board = d1_mini
framework = arduino

izbushka, 2021-07-21:
Similar issue is fixed now on FastLED library if it helps )
https://github.com/FastLED/FastLED/issues/1237

jdotedot, 2023-02-28:
I have a working Arduino project using WS2812B LEDs and the following configuration:
Board = ESP8266 version 3.0.2 - Lolin(WEMOS) D1 R2 & mini
Library = Adafruit NeoPixel version 1.11.0

When I upgrade the ESP8266 board to version 3.1.1, the project compiles, but the LED pixels are set incorrectly.  It looks like a timing problem.